TixiaoShan/LIO-SAM #1: hi, thanks for sharing! I just ran and had some problems with lio-sam

hunkyu, 2020-07-03:
The data set I used is caual_walk.bag and it looks like the location is not updated
[image]

YuehChuan, 2020-07-03:
I use it without problem. 
Great job, Sweet!

YuehChuan, 2020-07-03:
you need to install 
sudo apt install libparmetis-dev
use caual_walk.bag

terminal#1 **roscore**
terminal#2 **rosbag play --clock -l casual_walk.bag** 

terminal#3 **roslaunch lio_sam run.launch**
**Enjoy!**

TixiaoShan, 2020-07-03:
> you need to install
> sudo apt install libparmetis-dev
> use caual_walk.bag
> 
> terminal#1 **roscore**
> terminal#2 **rosbag play --clock -l casual_walk.bag**
> 
> terminal#3 **roslaunch lio_sam run.launch**
> **Enjoy!**

@YuehChuan 

I am curious why just playing the bag doesn’t work.

YuehChuan, 2020-07-03:
@TixiaoShan  I thought the bag will come to the end soon, It fact we should do it in speed run.
 I just want to reply  hunkyu how I made it.  Fixed.  Thanks for remind!   
terminal#1 **roscore**
terminal#2 **rosbag play casual_walk.bag -r 3**
terminal#3 **roslaunch lio_sam run.launch**

hunkyu, 2020-07-03:
I just made rosbag *.bag -r 3 according to your method, but still the same, I don’t know why! I use ubuntu 18.04. The develop branch used by gtsam @TixiaoShan @YuehChuan

hunkyu, 2020-07-03:
[image]
I just made rosbag *.bag -r 3 according to your method, but still the same, I don’t know why! I use ubuntu 18.04. The develop branch used by gtsam @TixiaoShan @YuehChuan

TixiaoShan, 2020-07-03:
@hunkyu 
It’s a gtsam problem. Have you tried the version mentioned in the readme?

YuehChuan, 2020-07-03:
@hunkyu I encountered this issue at first, too,  could you check whether you already install
 **sudo apt install libparmetis-dev** ?

hunkyu, 2020-07-04:
@TixiaoShan Thank you! It is indeed a gtsam version problem

closed 2020-07-04 by hunkyu