TixiaoShan/LIO-SAM #2: Problems with other dataset

getupgetup, 2020-07-03:
Thank your for your great job! Since LIO-SAM and [LIO-mapping](https://github.com/hyye/lio-mapping) both use lidar and imu data to run slam. I tested your code with dataset(slow2.bag) provided by LIO-mapping and tested LIO-mapping with your dataset(casual_walk.bag). Though I modified extrinsic rotation and translation, both of two tests failed(generating meaningless map). What's the problem? Is it because your imu data is used in a different way?

zwmisdu, 2020-07-03:
I have the similar problem.The bag(casual_wakl.bag) can perform well  when running LIO-SAM, but failed when i want to compare the result when running LEGO-LOAM. I have tried to use just  /points_raw when running LEGO but also failed.

hunkyu, 2020-07-03:
@zwmisdu What I am showing now is that it cannot be run in casual_wake.bag, and it fails. May I ask how you set it up and what parameters need to be changed

getupgetup, 2020-07-03:
> @zwmisdu What I am showing now is that it cannot be run in casual_wake.bag, and it fails. May I ask how you set it up and what parameters need to be changed

lio-sam performs well with casual_walk.bag. Use "git pull" to make sure your code is newest. You don't need to change anything.

zwmisdu, 2020-07-03:
> @zwmisdu What I am showing now is that it cannot be run in casual_wake.bag, and it fails. May I ask how you set it up and what parameters need to be changed

I have saw your problem. I can run it successfully without any change.

zwmisdu, 2020-07-03:
> @zwmisdu What I am showing now is that it cannot be run in casual_wake.bag, and it fails. May I ask how you set it up and what parameters need to be changed

just like the guy above said, i did not change anything

TixiaoShan, 2020-07-03:
> I have the similar problem.The bag(casual_wakl.bag) can perform well when running LIO-SAM, but failed when i want to compare the result when running LEGO-LOAM. I have tried to use just /points_raw when running LEGO but also failed.

LEGO won’t work with all the dataset provided here. It needs the lidar to be placed horizontally and a ground in the scan all the time.

TixiaoShan, 2020-07-03:
> Thank your for your great job! Since LIO-SAM and [LIO-mapping](https://github.com/hyye/lio-mapping) both use lidar and imu data to run slam. I tested your code with dataset(slow2.bag) provided by [...]

@getupgetup It’s highly likely an IMU extrinsic problem.

TixiaoShan, 2020-07-03:
@getupgetup 
Found the problem. The bags from LIO-mapping uses a 6-axis IMU. As mentioned in the readme, LIO-SAM only works with a 9-axis IMU currently.

dustinksi, 2020-07-04:
> @zwmisdu What I am showing now is that it cannot be run in casual_wake.bag, and it fails. May I ask how you set it up and what parameters need to be changed

Maybe your GTSAM library was not install correctly, I try to rebuild GTSAM with author's method described in README, then work great on casual_wakl.bag.

hunkyu, 2020-07-04:
@zwmisdu  Yes, it has been solved

getupgetup, 2020-07-06:
> @getupgetup
> Found the problem. The bags from LIO-mapping uses a 6-axis IMU. As mentioned in the readme, LIO-SAM only works with a 9-axis IMU currently.

It might be part of reason. Since LIO-mapping can't run well with casual_walk.bag, I think there'are some other reasons(extrinsic calibration maybe). Would you please have a look at LIO-mapping and find the difference between them if you have time? Thank you!

TixiaoShan, 2020-07-06:
@TixiaoShan 
Most of my dataset features aggressive motion, without compensation from gimbal or car. So other methods may struggle with it. I think part of the reason is because of the optimization pipeline of lio-mapping, which is adapted from VINS-Mono. I ran [a dataset](https://youtu.be/mBCJYiijxPM) with [calibrated extrinsic](https://github.com/ethz-asl/kalibr), the result of VINS-Mono is poor also.

getupgetup, 2020-07-07:
> @getupgetup
> Most of my dataset features aggressive motion, without compensation from gimbal or car. So other methods may struggle with it. I think part of the reason is because of the optimization pipeline of lio-mapping, which is adapted from VINS-Mono. I ran [a dataset](https://www.youtube.com/watch?v=yKIVOkjycGo) with [calibrated extrinsic](https://github.com/ethz-asl/kalibr), the result of VINS-Mono is poor also.

Thanks! By the way, is there anyway to generate IMU orientation data manually if my IMU data is only 6-axis? What did you do to get the orientation data?

TixiaoShan, 2020-07-07:
@getupgetup 

I don't think there is such a way to do that. My IMU outputs orientation data by default.

getupgetup, 2020-07-08:
> @getupgetup
> 
> I don't think there is such a way to do that. My IMU outputs orientation data by default.

Thanks~

closed 2020-07-08 by getupgetup