TixiaoShan/LIO-SAM #3: How to save map?

hitbg-yjm, 2020-07-03:
Hi, I  edit params.yaml and change   savePCD: true.
But it doesn't work .
the code:
ros::Rate rate(0.2);
        while (ros::ok()){
            rate.sleep();
            publishGlobalMap();       
        }

        if (savePCD == false)
            return;

when  ros::ok() wiil shun down?
Thanks!

TixiaoShan, 2020-07-03:
@hitbg-yjm 

- [I just made a new push for this problem.](https://github.com/TixiaoShan/LIO-SAM/commit/64a10e8e996ffee7154f144dbdeb6e2425dea9b9)
- 
- You also need to use ```sudo gedit /opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/nodeprocess.py``` to change the timeout threshold for killing the node. ROS will kill the node after ```_TIMEOUT_SIGINT``` seconds. A big map might needs a few minutes to save. Change it to a larger value if needed.

hitbg-yjm, 2020-07-08:
It helps a lot.

closed 2020-07-08 by hitbg-yjm