TixiaoShan/LIO-SAM #4: Evaluation using KITTI Vision Benchmark

hlezki, 2020-07-03:
Hi,

Thank you sharing your work. 
Is it possible to evaluate LIO-SAM on KITTI Odometry Dataset? Can you explain how to make it possible ?

Thanks in advance.

TixiaoShan, 2020-07-03:
@hlezki 
I am taking a look at this right now. I may come back later with the tutorial.

hlezki, 2020-07-03:
Thank you, I am looking forward to your explanation.

TixiaoShan, 2020-07-03:
@hlezki 

I have one question. Have you worked with KITTI IMU data? It seems that the IMU frequency is only at 1Hz.

hlezki, 2020-07-03:
No, I have not worked before with IMU data of KITTI Benchmark.

hunkyu, 2020-07-04:
@TixiaoShan The frequency given by kitti gps/imu is very low, there is no way to verify LIO-SAM

TixiaoShan, 2020-07-04:
@TixiaoShan 
Thanks for the information. I notice that the synced IMU frequency is the same as the lidar frequency. There is no way to integrate one IMU data during a scan for graph optimization in LIO-SAM.

daydayupzhang, 2020-07-07:
@TixiaoShan 
The IMU frequency in raw data/synchronized is very low, but in raw data/unsychronized it is 100Hz.
There is a blog about KITTI dataset, which especially mentions how to deal with the problems of IMU data. I hope this will help.
 https://zhuanlan.zhihu.com/p/75672946

closed 2020-07-11 by TixiaoShan

hlezki, 2020-07-12:
Thanks for your time.