TixiaoShan/LIO-SAM #5: What are the benefits of multiple updates in isam?

hunkyu, 2020-07-04:
In closed-loop correction, use isam->update() 5 times. Generally, we only use it once. Do you think it will be better to use it multiple times? But is it more time-consuming?

closed 2020-07-04 by hunkyu

narutojxl, 2020-07-04:
Hi @hunkyu, do you figure out the reason?  
5 times update in my computer Intel® Core™ i7-8700 CPU @ 3.20GHz × 12, will lead to the trajectory choppy like this picture. There is  a little large delay, although only one update. I test with `rosbag play casual_walk.bag  -r 1`
[image]

hunkyu, 2020-07-04:
@narutojxl  isam update can speed up convergence many times, I feel that in your case, more times isam update is okay

narutojxl, 2020-07-04:
Thanks you @hunkyu.

TixiaoShan, 2020-07-04:
@narutojxl @hunkyu 
The update time is really a personal preference for me. 5 times will help the isam converge faster but may cause a performance problem.

narutojxl, 2020-07-04:
* The large delay phenomenon is because default rviz config show `/lio_sam/mapping/cloud_registered` and `/lio_sam/mapping/map_local`, do not show this two topic.
* map frame and odom frame is the same frame, in code we can comment ` tfMap2Odom.sendTransform(tf::StampedTransform(map_to_odom, thisImu.header.stamp, "map", "odom"));` in `imuHandler()` of imuPreintegration.cpp, and then in rviz, we set the odom is the Global fixed frame. 
* About the oscillation in above picture. I found it only occurs in the xy plane. I try to comment the repetitive `isam->update()` from line 579  to line 582, and  from line 1265 to line 1269, floowing author saied above. The oscillation is moderate, but still exist. Do you know the reason @TixiaoShan and @hunkyu ?  Thanks a lot!