TixiaoShan/LIO-SAM #6: extrinsicRPY meaning in readme

narutojxl, 2020-07-04:
Hi @TixiaoShan, thanks for your excellent work !
when reading about `extrinsicRPY` in doc and  `imuConverter()` function, i don't understand this param's meaning.
Imu's orientation is `q_b0_bk`, which b0 is the imu's world frame, b0 imu frame in its world frame in time stamp  k.
`extrinsicRot` is q_l2b, different from `extrinsicRPY`.  What's the meaning of `q_b0_bk * extQRPY`? 
It isn't  `q_b0_lk = q_b0_bk * q_l2b.traspose()`, nor `q_l0_lk = q_l2b * q_b0_bk * q_l2b.traspose()`, right?

TixiaoShan, 2020-07-04:
@narutojxl 
The reason there is two extrinsic rotation params is that my IMU's inertial reading (acc and gyro) frame is different from the orientation frame. The pitch of my IMU is rotating around IMU-x. The pitch of the lidar is rotating around Lidar-y. Thus I use extrinsicRPY to rotate IMU orientation -90 degrees to the lidar frame, so they can be consistent.

narutojxl, 2020-07-04:
Hi @TixiaoShan,
Sorry, i don't look the  frame picture carefully,  thank you vey very much :)   My problem is solved, closed this issue.
BTW,  your imu's yaw+ arrow direction is different from roll+ and pitch+.

closed 2020-07-04 by narutojxl

TixiaoShan, 2020-07-04:
@narutojxl 
It is indeed. I guess it's a preference of the manufacturer. That took me a while to figure out and to get the correct results.

KitKat7, 2020-07-22:
> It is indeed. I guess it's a preference of the manufacturer. That took me a while to figure out and to get the correct results.

Hi @TixiaoShan, I found the yellow arrow indicating the IMU-z rotation in the current https://github.com/TixiaoShan/LIO-SAM/blob/master/config/doc/imu-transform.png was left-handed, while in the 3DM-GX5-25's user manual, it was right-handed. Is it because of the IMU setting differences?

chennuo0125-HIT, 2020-12-23:
no need config param "extrinsicRPY", because imu orientation wont be used in any function

valgur, 2020-12-28:
@chennuo0125-HIT That's not true, though. IMU orientation is converted into `imuRollInit`, `imuPitchInit` and `imuYawInit` and published with the `cloud_info` message and then used in quite a few places in `mapOptimization.cpp`.

chennuo0125-HIT, 2020-12-29:
@valgur yes, you are right , i didnt saw cloud_info carefully :(

Cristian-wp, 2022-05-04:
Hello, I can not understand very well the meaning of this matrix.
For what I have understand and tested, the extrinsicRPY have the purpose to aling the LIDAR RPY to the IMU ones. 
So it can be a dummy question, but if my IMU RPY are different for the one used in the youtube debug imu, the signs must be coerent with mine not with the video one, right?

TZECHIN6, 2023-10-27:
I have question about using extrinsicRPY and the urdf file. Should I still need to input any transform in the urdf? as the param file already take care of the imu transformation, if I input again in the urdf, will it duplicate the transformation and causing unexpected result?

```
 <link name="imu_link"> </link>
  <joint name="imu_joint" type="fixed">
    <parent link="chassis_link" />
    <child link="imu_link" />
    <origin xyz="0 0 0" rpy="0 0 0" />
  </joint>
```