TixiaoShan/LIO-SAM #7: updateInitialGuess()

narutojxl, 2020-07-04:
Hi @TixiaoShan,
When get the current laser's initialize pose in map, i debug this code in `updateInitialGuess()` function in `mapOptmization.cpp`:
> 
     if (cloudInfo.imuAvailable == true) 
        {   
           //this line is me added 
            ROS_WARN("odom = %d, cloud = %d, self = %d ", cloudInfo.odomAvailable, cloudInfo.imuPreintegrationResetId , imuPreintegrationResetId);

            Eigen::Affine3f transBack = pcl::getTransformation(0, 0, 0, cloudInfo.imuRollInit, cloudInfo.imuPitchInit, cloudInfo.imuYawInit);
            Eigen::Affine3f transIncre = lastImuTransformation.inverse() * transBack;

            Eigen::Affine3f transTobe = trans2Affine3f(transformTobeMapped);
            Eigen::Affine3f transFinal = transTobe * transIncre;
            pcl::getTranslationAndEulerAngles(transFinal, transformTobeMapped[3], transformTobeMapped[4], transformTobeMapped[5], 
                                                          transformTobeMapped[0], transformTobeMapped[1], transformTobeMapped[2]);

            lastImuTransformation = pcl::getTransformation(0, 0, 0, cloudInfo.imuRollInit, cloudInfo.imuPitchInit, cloudInfo.imuYawInit); // save imu before return;
            return;
        }
The output is:
> 
	process[lio_sam_rviz-8]: started with pid [14288]
	[ INFO] [1593869689.438855893]: ----> IMU Preintegration Started.
	[ INFO] [1593869689.471981403]: ----> Image Projection Started.
	[ INFO] [1593869689.498746662]: ----> Feature Extraction Started.
	[ INFO] [1593869689.507955480]: ----> Map Optimization Started.
	[ WARN] [1593869693.421098667]: odom = 0, cloud = 0, self = 0 
	[ WARN] [1593869693.626720210]: odom = 0, cloud = 0, self = 0 
	[ WARN] [1593869869.932765664]: odom = 1, cloud = 0, self = 1 
	[ WARN] [1593869870.131129145]: odom = 0, cloud = 0, self = 1 
	[ WARN] [1593869874.965915153]: odom = 1, cloud = 1, self = 2 
	[ WARN] [1593869875.179717063]: odom = 0, cloud = 1, self = 2 
	[ WARN] [1593869875.376471333]: odom = 0, cloud = 1, self = 2 
	[ WARN] [1593869905.031495240]: odom = 1, cloud = 2, self = 3 
	[ WARN] [1593869905.231531192]: odom = 0, cloud = 2, self = 3 
	[ WARN] [1593869909.896946401]: odom = 1, cloud = 3, self = 4 
	[ WARN] [1593869910.136724098]: odom = 1, cloud = 3, self = 4 
	[ WARN] [1593869910.275975347]: odom = 0, cloud = 3, self = 4 
	[ WARN] [1593869919.954947659]: odom = 1, cloud = 4, self = 5 
	[ WARN] [1593869920.157889474]: odom = 0, cloud = 4, self = 5 
	[ WARN] [1593869925.202043337]: odom = 1, cloud = 5, self = 6 
	[ WARN] [1593869925.428638865]: odom = 0, cloud = 5, self = 6 
	[ WARN] [1593869930.039206332]: odom = 1, cloud = 6, self = 7 
	[ WARN] [1593869930.244444627]: odom = 1, cloud = 6, self = 7 
	[ WARN] [1593869935.081837640]: odom = 1, cloud = 7, self = 8 
	[ WARN] [1593869935.286080622]: odom = 0, cloud = 7, self = 8 
	[ WARN] [1593869944.967022261]: odom = 1, cloud = 8, self = 9 
	[ WARN] [1593869945.161093601]: odom = 0, cloud = 8, self = 9 
	[ WARN] [1593869950.010743418]: odom = 1, cloud = 9, self = 10 
	[ WARN] [1593869950.215691434]: odom = 1, cloud = 9, self = 10 
	[ WARN] [1593870259.647486738]: odom = 0, cloud = 10, self = 10 
	[ WARN] [1593870325.011413956]: odom = 1, cloud = 10, self = 11 
	[ WARN] [1593870325.214021085]: odom = 0, cloud = 10, self = 11 
	[ WARN] [1593870329.867247998]: odom = 1, cloud = 11, self = 12 
	[ WARN] [1593870330.052590294]: odom = 0, cloud = 11, self = 12 
	[ WARN] [1593870330.254257450]: odom = 0, cloud = 11, self = 12 
	[ WARN] [1593870335.117750271]: odom = 1, cloud = 12, self = 13 
	[ WARN] [1593870335.295067667]: odom = 0, cloud = 12, self = 13 
	[ WARN] [1593870335.505050773]: odom = 0, cloud = 12, self = 13 
	[ WARN] [1593870344.980313780]: odom = 1, cloud = 13, self = 14 
	[ WARN] [1593870345.181445314]: odom = 0, cloud = 13, self = 14 
	[ WARN] [1593870345.382959890]: odom = 0, cloud = 13, self = 14 
	^C[lio_sam_rviz-8] killing on exit
	[navsat-7] killing on exit
	[ekf_gps-6] killing on exit
	[lio_sam_mapOptmization-5] killing on exit
	[lio_sam_featureExtraction-4] killing on exit
	[lio_sam_imageProjection-3] killing on exit
	[lio_sam_imuPreintegration-2] killing on exit
	****************************************************
	Saving map to pcd files ...
	Processing feature cloud 956 of 957 ...****************************************************
	Saving map to pcd files completed
	[rosout-1] killing on exit
	[master] killing on exit
	shutting down processing monitor...
	... shutting down processing monitor complete
	done
The `cloudInfo.odomAvailable` in each msg is not always true when after initialized, is it normal? 
Thanks for your help!

TixiaoShan, 2020-07-04:
It's normal. As you can see, ```cloudInfo.imuPreintegrationResetId``` is changed multiple times. I assume the graph is reset due to loop closure. Then the odometry from imuPreintegration will be teleported and cannot be used for initial guess.

narutojxl, 2020-07-06:
Thanks a lot !

closed 2020-07-06 by narutojxl