TixiaoShan/LIO-SAM #9: How about performance of a low cost imu ?

Oofs, 2020-07-06:
Thanks for your great job. It's really awesome.
I have googled that 3DM-GX5-25 costs about [$1600](https://www.lord.com/content/3dm-gx5-25-ahrs), which is a high performance IMU. So, I'm curious about the importance of the inertial measurement accuracy in your algorithm, will a consumer grade IMU do the same job ?

TixiaoShan, 2020-07-06:
@Oofs 
I tested LIO-SAM on a low-cost IMU (<$100) with 75Hz. The results are ok. With a low-frequency IMU, the results are only good when the playback speed is "-r 1".
https://github.com/TixiaoShan/Stevens-VLP16-Dataset

Oofs, 2020-07-13:
Thanks, I will have a try!

closed 2020-07-28 by stale[bot]

horverno, 2021-01-28:
I am about to buy an IMU, is there maybe a list of recommended IMUs? I don't want to buy a low cost one but neither spend $1600; so a halfway option would be the best, possibly a 200Hz+ 9-axis model. thanks
ps @TixiaoShan thanks for this good work :+1: