TixiaoShan/LIO-SAM #10: when it would be available with Ouster lidar? what is the main issue?

JustWon, 2020-07-07:
I am interested in using this with Ouster lidar sensor.
Could I directly plug the Ouster sensor or 
are there any issues to consider?

Thanks for this great framework!

zwmisdu, 2020-07-07:
> I am interested in using this with Ouster lidar sensor.
> Could I directly plug the Ouster sensor or
> are there any issues to consider?
> 
> Thanks for this great framework!

I also have this question. At now, data from Ouster can not be used directly. The author has said that supporting OusterLidar would be the next work to do. 
In general, the problems need to be sloved are the format of scan data.If there is something can be done together, I am always glad to  lend a hand.

TixiaoShan, 2020-07-07:
I have ordered OS1 gen2 since February and received it last month, there is no official driver for it yet.

zwmisdu, 2020-07-07:
> I have ordered OS1 gen2 since February and received it last month, there is no official driver for it yet.

I can send the file to you. My LIDAR is  Ouster64,and I couldn't find your email, maybe you can give me the address.

TixiaoShan, 2020-07-07:
> > I have ordered OS1 gen2 since February and received it last month, there is no official driver for it yet.
> 
> I can send the file to you. My LIDAR is Ouster64,and I couldn't find your email, maybe you can give me the address.

Thanks for offering it. I have os64 bags. I am just waiting for the driver for gen2 so both generations can be supported later.

zwmisdu, 2020-07-07:
> Thanks for offering it. I have os64 bags. I am just waiting for the driver for gen2 so both generations can be supported later.

Okay.looking forward to your excellent work! And I will try to modify the code to be useful for my lidar data. By the way, I should pay attention to the channel of 'ring'and ‘channel’ in Velodyne Lidar and obtain the corresponding parameter in Ouster(if there are exactly corresponding channel will be great), is that right?

TixiaoShan, 2020-07-07:
> Okay.looking forward to your excellent work! And I will try to modify the code to be useful for my lidar data. By the way, I should pay attention to the channel of 'ring'and ‘channel’ in Velodyne [...]

In velodyne, the two channels are "ring" and "time". Something similar should exist in Ouster cloud. You can simply "rostopic echo" the point cloud topic and see what's in the fields. For your lidar, "ring" should vary from 0 to 63. The timestamp of a point in Ouster is called "t" if I remember correctly.

TixiaoShan, 2020-07-07:
@JustWon @zwmisdu 
I am updating the repo for Ouster lidar today. Stay tuned.

TixiaoShan, 2020-07-07:
Ouster lidar is now supported.

closed 2020-07-07 by TixiaoShan