TixiaoShan/LIO-SAM #12: Performance when sensors are relatively fast

henry0572, 2020-07-09:
Hello!
Thank you for a great job.

I was looking at the LIO-SAM code, and the following commented out section caught my attention.

https://github.com/TixiaoShan/LIO-SAM/blob/master/src/imageProjection.cpp#L450-L459
```
        // If the sensor moves relatively slow, like walking speed, positional deskew seems to have little benefits. Thus code below is commented.

        // if (cloudInfo.odomAvailable == false || odomDeskewFlag == false)
        //     return;

        // float ratio = relTime / (timeScanNext - timeScanCur);

        // *posXCur = ratio * odomIncreX;
        // *posYCur = ratio * odomIncreY;
        // *posZCur = ratio * odomIncreZ;
 ```

What were the results when sensors are relatively fast and you uncommented above?
Also, is there a dataset that I can try?

TixiaoShan, 2020-07-09:
I don't have a dataset that has a very fast speed. The maximum speed for my dataset is 6km/h. So the difference if pretty small when it is commented. I image those lines will be very useful for datasets gathered by a car.

henry0572, 2020-07-11:
I see.
Thanks for the reply.

closed 2020-07-11 by henry0572

ellieko, 2023-12-03:
@TixiaoShan Hello, thanks for sharing this package! I'm trying out this package for a car that is moving with a fast speed (around 40-50mph) but it doesn't... work well. Do you have any advice on parameter setting? I changed some params in params.yaml, including extrinsicRot, and commented out the condition, if (vel.norm() > 30)..., in imuPreintegration.cpp, but its point cloud map gets disconnected after a few seconds of the bag. Any advice would be really appreciated!