TixiaoShan/LIO-SAM #15: IMU rate for calculating discrete-time noise parameter

wwtinwhu, 2020-07-11:
Hello，Dr. Shan. Excellent Work!  With respect to imu preintegration,  you use gtsam's implementation. The question is that  where can we  offer the imu rate for calculating discrete-time noise parameter? I didn't find the imu rate setup in your code.  And I assume the imu nosie parameter in params.yaml is in continuous-time. Am I right? Thank you very much

TixiaoShan, 2020-07-11:
IMU rate doesn't need to be provided. GTSAM will integrate all the available measurements. The imu noise in params is calibrated using Kalibr when I put IMU static for at least 2 hours.

wwtinwhu, 2020-07-11:
> IMU rate doesn't need to be provided. GTSAM will integrate all the available measurements. The imu noise in params is calibrated using Kalibr when I put IMU static for at least 2 hours.

Thanks!

closed 2020-07-11 by wwtinwhu

wwtinwhu, 2020-07-13:
For this issue, I find that in the GTSAM class PreintegratedImuMeasurements , the method called integrateMeasurement have a parameter called dt, which allows to pass from continuous time noise to discrete time noise.  dt can be calculated liked the imuPreintegration.cpp line 357. So IMU rate doesn't need to be provided.