TixiaoShan/LIO-SAM #16: Extrinsic Calibration

hlezki, 2020-07-12:
Hi,

Thanks again for your great work. 
Can you explain how did you obtain extrinsic parameters between Lidar and IMU?
I need to obtain extrisics between Lidar and IMU for a ground vehicle such as car. Lidar is placed at the top of the car and IMU is placed at the luggage of the car. It is not possible to move the car arround all three axes and along all three axes (the motion is approximately planar). So, hand-eye calibration fails for the height between lidar and IMU. Can you recommend any tool or code that provides extrinsic calibration between IMU and Lidar ?

 Thanks in advance.

TixiaoShan, 2020-07-13:
I don't have experience calibrating sensors on a full-sized car. It seems that your setup is similar to the KITTI sensor suite.  Have you checked their calibration process?

hlezki, 2020-07-13:
Yes, I checked KITTI's calibration process. They use hand-eye calibration for Lidar-IMU extrinsic parameters. But hand-eye calibration can not provide exact 6-DOF calibration if you are not moving vehicle arround all three axes. 

Thanks for the reply

closed 2020-07-28 by stale[bot]

Pallav1299, 2020-08-26:
@hlezki Did you try finding the Lidar-IMU extrinsic parameters? How was the result? 
Also, can you suggest some resources/packages for the same?

hlezki, 2020-08-27:
Hi @Pallav1299 

You can try Lidar Align (https://github.com/ethz-asl/lidar_align). However, it does not realistic enough for vehicles that move on a 2D plane like cars. You can also check the following link where some useful calibration tools are listed. 

https://github.com/ethz-asl/mav_tools_public/wiki/Calibration 

However, I could not find any ready-to-use tool for cars. The given methods require moving the vehicle arround all three axes that is not possible for cars.

Romansp12, 2021-03-15:
@Pallav1299 @hlezki 

Hello, I see that you were asking the same questions as I am now

Could you tell us how you solved this problem at the end, calibrate the lidar to the IMU on the machine?
More precisely, I need to calibrate the lidar to base_link (but this is the same as calibrating it to the IMU). Perhaps this can be done without an IMU, in other ways? maybe you know modern libraries / packages for this