TixiaoShan/LIO-SAM #19: The error result for Campus dataset (large) dataset

sysuzyc, 2020-07-14:
Hi, @TixiaoShan, thanks for your excellent work !
There are some questions for Campus dataset (large) dataset.  I have changed the `imuTopic` to `imu_correct` and `extrinsicRot`, `extrinsicRPY` to Identity matrix in `config/params.yaml`. 
And the rosbag plays at normal speed:`rosbag play big_loop.bag` 
But the final result is not as expected.  There is a big shift in the global map.
Here is the result in my computer.
[image]
And this is the demo result in LIO-SAM Paper.
[image]
The red and black block is the different area.
The detail is shown as:
[image]
[image]

How can I get the correct result? 
Thanks very much!

TixiaoShan, 2020-07-14:
That's expected because LIO-SAM has drift when loop closure is not enabled.

TixiaoShan, 2020-07-14:
You can enable the loop closure function by following README instructions.
https://github.com/TixiaoShan/LIO-SAM/issues/21

sysuzyc, 2020-07-14:
Thanks. I have changed the `loopClosureEnableFlag` parameter to `true` in `config/params.yaml` . 
The result is better now.
[image]
[image]

**By the way**, the IMU used in this project is 9-axis. The orientation could be get defaultly.  If there is a 6-axis IMU, could I get the orientation by Integrating the angular velocity? Therefore, the 6-axis IMU could also be used in this project.

TixiaoShan, 2020-07-14:
Theoretically, it's possible to use a 6-axis IMU. There are only a few places used roll, pitch, and yaw info.

sysuzyc, 2020-07-15:
OK. I will try it. Thanks very much.

closed 2020-07-15 by sysuzyc