TixiaoShan/LIO-SAM #20: Why Large Velocity Happens - Large velocity, reset IMU-preintegration!?

carlin314, 2020-07-14:
Hi,

Thx for great work, I use our dataset and find imu path is not that stable and the imu when turn happens, alway like zigzag, 

and when in large yawrate, failure happens in so big velocity, any ideas?

[image]

BR

TixiaoShan, 2020-07-14:
It's probably the extrinsic between IMU and Lidar is not correct, so the imu integration is wrong.

carlin314, 2020-07-15:
> It's probably the extrinsic between IMU and Lidar is not correct, so the imu integration is wrong.

Thx, the extrinsics are right, I find it is the imu data which is not right for it, it took me a while to get that right to work

carlin314, 2020-07-15:
If someone also has imu data problem, Mine is under as a reference, mine is Novatel pp7d-e1, which inertial is EG320N
[image]

closed 2020-07-15 by carlin314

zwmisdu, 2020-07-20:
> If someone also has imu data problem, Mine is under as a reference, mine is Novatel pp7d-e1, which inertial is EG320N
> [image]

I have the same problem as yours. Could you leave the e-mail so that we can communicate further?

ibrahimhroob, 2021-04-27:
Hi @zwmisdu 

I am having the same issue, have you figured out how to solve it?