TixiaoShan/LIO-SAM #21: Run big-loop.bag data set with different results

windwolf9, 2020-07-14:
Hi,@TixiaoShan,I run big-loop.bag data set, the results of  loop closure is different，and Sometimes the results are good, sometimes not!
[20200714181842](https://user-images.githubusercontent.com/17972452/87414531-801d7280-c5fe-11ea-8c84-15eeafe8b4c5.png)

[image]

TixiaoShan, 2020-07-14:
The loop closure results come from PCL icp. Sometimes ICP gives bad results. A more advanced loop closure can be found here.
https://github.com/irapkaist/SC-LeGO-LOAM

windwolf9, 2020-07-15:
Thanks,and I found that the loop closure is good In the beginning()，and it will going to get worse immediately and  an error occurs :[ WARN] [1594800092.012221229]: Large velocity, reset IMU-preintegration! When it happened，the maps got worse。

The results are as follows：
[image]
[image]
[image]

TixiaoShan, 2020-07-15:
The current implementation of loop closure is not ideal. The success rate heavily depends on the performance of PCL icp.

TixiaoShan, 2020-07-15:
**A better implementation is welcome from the community.** 
**Feel free to join and discuss how to improve its performance.**

TixiaoShan, 2020-07-27:
> **A better implementation is welcome from the community.**
> **Feel free to join and discuss how to improve its performance.**

Loop closure should be fixed now.

closed 2020-07-27 by TixiaoShan