TixiaoShan/LIO-SAM #22: how to use gps to evaluation

limeng1523, 2020-07-14:
Hi,
I want to use the gps provided by your data sets as the ground truth for evaluation, but its coordinate system and frequency are different from the system output, could you tell me how do you solve this? Thank you very much!

TixiaoShan, 2020-07-14:
When you launch the system, there is one odometry published at ```odometry/gps```, which is the transformed GPS coordinates in Cartesian space. You can use that for evaluation.

limeng1523, 2020-07-15:
@TixiaoShan Hi,
Thanks for your answer. I have found the topic /odometry/gps, but it can't coincide with the system output. It seems that there is a rotation between /odometry/gps and system output. Besides, during running the system, the gps position changed suddenly. Could you tell me where is the problem? Thank you very much!
[image]
[image]

TixiaoShan, 2020-07-15:
You can set the ```gpsTopic``` correctly to subscribe to it. The GPS data in the dataset is very bad. It jumps a few hundred meters will corrupt the optimization.

TixiaoShan, 2020-07-15:
https://github.com/TixiaoShan/LIO-SAM/issues/8

closed 2020-07-30 by stale[bot]

Pallav1299, 2020-08-03:
I tried the package on **Campus dataset(small)**. When not using GPS, the package performs very well. 
Some results:
[image]
[image]

On comparing the GPS odometry with the optimized path, both look quite similar. The initial rotation seems to be the only difference between the two.

But when using GPS, there are some serious artifacts in the output PCD. 
Some results:
[image]

Now I have the following questions:
1. Can we create correct(with least artifacts) global PCD map with LIO-SAM? 
2. Any method for evaluating LIO-SAM? (I tried evaluating with KITTI ground truth sequences, the elevation data was inconsistent: [LINK)](https://github.com/TixiaoShan/LIO-SAM/issues/35#issuecomment-664587989. But this can be misleading since we don't have correct IMU parameters for KITTI.)
3. How well does LIO-SAM support car-like vehicles?