TixiaoShan/LIO-SAM #29: [Question] Accuracy and method for LIDAR-IMU extrinsic calibration

YoshuaNava, 2020-07-14:
Hi Tixiao,
The results from this package look very impressive.

By means of this issue I would like to ask you the following three questions:

1) What's the expected accuracy of the LIDAR->IMU transformation?

2) How do you estimate it? Do you use CAD or an online estimation package?

3) Can this package improve upon an inaccurate calibration over time_

TixiaoShan, 2020-07-15:
```
What's the expected accuracy of the LIDAR->IMU transformation?
```
The more accurate the better. But the system is robust enough to handle some misalignment.

```
How do you estimate it? Do you use CAD or an online estimation package?
```
Simply hand-eye calibration for my setup as the IMU is right close to lidar.

```
Can this package improve upon an inaccurate calibration over time?
```
No online calibration is implemented.

closed 2020-07-29 by YoshuaNava

SubMishMar, 2020-10-28:
Hi, Thanks for sharing your work. Is there a package you recommend to do Hand Eye Calibration of lidar and IMU? If not then can you kindly give some general inputs on how you did it?
Thanks.