TixiaoShan/LIO-SAM #30: Question about Parameter extrinsicTrans  and transformation betwen imu and lidar

wwtinwhu, 2020-07-15:
Hi, Tixiao.
      From the picture about lidar and imu coordinates from readme,  it is shown that the translation bewtween the origin of lidar and imu is a few centimeters. My question is why you set [0,0,0] for the parameter extrinsicTrans. 
      Second question is about the formula which transform imu pose to lidar pose, where in imuPreintegration.cpp line 74,75:
`gtsam::Pose3 lidar2Imu = gtsam::Pose3(gtsam::Rot3(1, 0, 0, 0), gtsam::Point3(extTrans.x(), extTrans.y(), extTrans.z()));;`  In your code, you firstly use imuConverter() to align the axis of the two coordinates, and then use lidar2Imu or imu2Lidar to get that.  I use this formula to transform imupose to lidarpose, where the rotation matrix of T_imu_lidar is identity matrix. 
T_lidar0_lidarn = T_lidar_imu * T_imu0_imun * T_imu_lidar
And this is different from the imu2Lidar in your code. Please give me some advice! Thank you.

TixiaoShan, 2020-07-15:
```
My question is why you set [0,0,0] for the parameter extrinsicTrans.
```
I changed the mounting position of the IMU a few times in different datasets. But the positions are always close to the lidar. So I didn't bother changing this param every time when I test different datasets. Strictly speaking, my approach is not ideal. This param needs to be provided for better performance.

After imuConverter(), the measurements of IMU should be in REP frame. So just a translation is needed. I am not sure I understand your question. Are you talking about using multiple imu and lidar here?

wwtinwhu, 2020-07-16:
> ```
> My question is why you set [0,0,0] for the parameter extrinsicTrans.
> ```
>
> I changed the mounting position of the IMU a few times in different datasets. But the positions are always close to the [...]

Thank you.  I get the your point about the first question.  Second question is hard to express. T_lidar0_lidarn means the transformation from time n lidar to time 0 lidar.

closed 2020-07-31 by stale[bot]