TixiaoShan/LIO-SAM #31: [Question] function projectPointCloud in imageProjection.cpp

minwo0jung, 2020-07-15:
Hello, I'm studying LIO-SAM and I intensively read that code.
Under is about my question.

Through publishClouds in imageProjection.cpp, cloudInfo, i.e., deskew point, is passed to featurExtraction.cpp. At this time, the cloudExtraction function gives the rangeMat.at<float>(i,j) value to cloudInfo.pointRange, which is the range value measured before removing the distortion of the point in the projectPointCloud function. After giving the range value, thisPoint is removed distortion using deskewPoint.

 In featureExtraction.cpp, smoothness is calculated using an distorted range value, and I want to know why diff is measured using distorted range instead of undistorted range.

TixiaoShan, 2020-07-15:
Thanks for pointing that out. The code is heavily adapted from LeGO-LOAM. I think that's a bug. I will push a new commit to fix it.

minwo0jung, 2020-07-15:
Thanks for your quick replying.
I think LIO-SAM is really good method for Lidar-SLAM. When I find another problem, I'll write again.

Thank you.

closed 2020-07-15 by minwo0jung

TixiaoShan, 2020-07-15:
@minwo0jung 
Thank you!