TixiaoShan/LIO-SAM #32: Some problems occurred during operation

eminbogen, 2020-07-15:
[image]
Hello. I used your public data to run your program, and this happened. The z coordinate keeps dropping. What is the problem? Is my configuration file not modified?

TixiaoShan, 2020-07-15:
Sounds like a configuration problem. I suggest testing with the latest code and the Casual Walking dataset.

eminbogen, 2020-07-16:
Thank you, I found the reason is that I forgot to modify extrinsicRot and extrinsicRPY. Your results are really good.

closed 2020-07-16 by TixiaoShan

duongthao1218, 2022-02-14:
> Thank you, I found the reason is that I forgot to modify extrinsicRot and extrinsicRPY. Your results are really good.

How to modify extrinsicRot and extrinsicRPY?