TixiaoShan/LIO-SAM #36: ICP compared with method of point matching against line or surface

Gaochao-hit, 2020-07-20:
I have noticed that in your LOOP_Closure you use the ICP method to calculate the transforamtion and in scan2MapOptimization() functionyou use method of point matching against line or surface. So I want to ask can the ICP in the LOOP_Closure be replaced by the method in scan2MapOptimization() ?

TixiaoShan, 2020-07-20:
It's definitely possible to replace it.

Gaochao-hit, 2020-07-20:
> It's definitely possible to replace it.

I mean what makes you choose ICP to detect loop, not the method of matching point to edge or surface. What the advantages and disadvantages of the above two methods? Does the ICP method have better convergeing ability? 
And I also notice that the fitneescore is used  to measure the reliability of ICP, but in my practice, the score can't measure the reliability well and I can't find a good method to measure the outcome until now. Can you give me some advice, Thank you.

TixiaoShan, 2020-07-23:
Choosing ICP is simply for easy implementation because it is adapted from LeGO-LOAM directly.

Gaochao-hit, 2020-07-26:
> Choosing ICP is simply for easy implementation because it is adapted from LeGO-LOAM directly.

Thank you for your reply and excellenr work.

closed 2020-07-26 by Gaochao-hit