TixiaoShan/LIO-SAM #37: How to project the SLAM map onto Google map?

nkliuhui, 2020-07-20:
I want to ask a question that is not relative to LIO-SAM. How to project the SLAM map onto Google map?

Thank you in advance!

TixiaoShan, 2020-07-20:
To do that, I use GIMP for pictures, Final Cut Pro for videos.

nkliuhui, 2020-07-20:
Emmm, Okey! (I want to make a visual interface, which can project slam map onto Google map in real time)

TixiaoShan, 2020-07-20:
I see. Then there is gonna be a lot of coding involved. I don't know Google earth or map API. But I can imagine you probably need to send some images to it for visualization.

nkliuhui, 2020-07-20:
Thank you for you suggestion. I'll try it!

kdaun, 2020-07-20:
For displaying satellite maps in rviz, take a look at [rviz_satellite](https://github.com/gareth-cross/rviz_satellite).

nkliuhui, 2020-07-20:
> For displaying satellite maps in rviz, take a look at [rviz_satellite](https://github.com/gareth-cross/rviz_satellite).

Thank you!

sam1992xjh, 2020-07-27:
Maybe you want this？
https://github.com/gareth-cross/rviz_satellite

closed 2020-07-27 by TixiaoShan

YangSiri, 2020-09-14:
Anyone knows how to visualize the real-time factor graph, like the connection between two state? Thank you~

cyborgdennett, 2022-06-07:
Hello guys, I made some adjustments to make [rviz_sattelite](https://github.com/gareth-cross/rviz_satellite) work with lio-sam

Here are the steps to reproduce.
## Step 1
install rviz_sattelite under same catkin_ws as lio-sam.

## Step 2
underneath /catkin_ws/src/[LIO-SAM/launch/include/config/rviz.rviz](https://github.com/TixiaoShan/LIO-SAM/blob/master/launch/include/config/rviz.rviz)

I added the following under ```Visualization Manager:```.

I tried with zoom 20, which did not work. But zoom 19 works. [osm zoom_levels](https://wiki.openstreetmap.org/wiki/Zoom_levels)

```
Class: ""
  Displays:
    - Alpha: 1
      Blocks: 3
      Class: rviz_plugins/AerialMapDisplay
      Draw Behind: true
      Enabled: true
      Name: AerialMapDisplay
      Object URI: https://tile.openstreetmap.org/${z}/${x}/${y}.png
      Topic: /gps/fix
      Value: true
      Zoom: 19
    - Class: rviz/Group
      Displays:
        - Class: rviz/TF
          Enabled: true
          Frame Timeout: 15
          Frames:
            All Enabled: true
            base_link:
              Value: true
            map:
              Value: true
          Marker Scale: 10
          Name: TF
          Show Arrows: true
          Show Axes: true
          Show Names: true
          Tree:
            map:
              base_link:
                {}
          Update Interval: 0
          Value: true
```

And changed the panels to:
```
Panels:
  - Class: rviz/Displays
    Help Height: 191
    Name: Displays
    Property Tree Widget:
      Expanded:
        - /AerialMapDisplay1
        - /links1
        - /Odometry1
        - /Odometry1/Odom GPS1
        - /Point Cloud1
        - /Feature Cloud1
        - /Mapping1
      Splitter Ratio: 0.5600000023841858
    Tree Height: 663
  - Class: rviz/Selection
    Name: Selection
  - Class: rviz/Views
    Expanded:
      - /Current View1
    Name: Views
    Splitter Ratio: 0.5
  - Class: rviz/Tool Properties
    Expanded: 
      - /2D Pose Estimate1
      - /2D Nav Goal1
      - /Publish Point1
    Name: Tool Properties
    Splitter Ratio: 0.5
  - Class: rviz/Time
    Experimental: false
    Name: Time
    SyncMode: 0
    SyncSource: "" ```

output: (I recommend putting alpha variable of AerialDisplay pretty low so that the PointMap is still visible
![image](https://user-images.githubusercontent.com/40475862/172404173-fc26e9cb-c07c-4a21-85d4-f8a4cda0fdba.png)

cyborgdennett, 2022-06-07:
Next step I want is: make a Path of the gps-coordinates so I can see the difference between the SLAM Path and gps-path. Anybody knows how to visualize the raw gps-coordinates into a Path?