TixiaoShan/LIO-SAM #42: Base_link problem?

yanlong658, 2020-07-26:
Thanks for your great job. It's so amazing
I use the original parameter in your params.yaml, and run the park.bag file. The rviz always show me some error on base_link. AS following the figure1. The ros system shows me error message as following figure2.
 
Bless you have the good day.

figure1.
[image]

figure2.
[image]

TixiaoShan, 2020-07-26:
It's a gtsam problem.
https://github.com/TixiaoShan/LIO-SAM/issues/1

yanlong658, 2020-07-27:
Thank you, I update my gtsam. It can run.

closed 2020-07-27 by TixiaoShan