TixiaoShan/LIO-SAM #46: Question about the number of the surface points.

Gaochao-hit, 2020-07-30:
I have noticed that the are so many surface points in the extracted features. Is that a bug ?

Gaochao-hit, 2020-07-30:
And I find that if I set curvature limit on the surface points, the number of surface points will decrease a lot and also the computation consumption will also decline a lot. At the same time, the accuracy will not decline.

Gaochao-hit, 2020-07-30:
And I find that if a set a curvature limit on the surface points, although the number of the extracted surface points is not very small, the odometry will drift very fast.

TixiaoShan, 2020-07-30:
I don't think it's a bug. If there are little features, the package won't work.

Gaochao-hit, 2020-07-30:
> I don't think it's a bug. If there are little features, the package won't work.

You are right. But the curvature of surface points extracted in the code can be very large and the number of surface points is also very large, which will increase computating load. when I set a threshold 0.1 to choose the surface points, there are about several hundreds of surface points which I think is sufficient for calculating transforamtion. But unfortunatelly, it fails and drift away quickly. This confuses me a lot.

closed 2020-08-01 by Gaochao-hit