TixiaoShan/LIO-SAM #49: poseCovariance

Gaochao-hit, 2020-08-02:
Shouldn't the covariance of the pose in the Mapping odometry increase constantly. I noticed that the poseCovariance sometimes just fluctuated.

TixiaoShan, 2020-08-02:
@Gaochao-hit 
I did notice that too. I don't know the reason behind it, probably something to do with the internal calculation of gtsam.

Gaochao-hit, 2020-08-03:
> @Gaochao-hit
> I did notice that too. I don't know the reason behind it, probably something to do with the internal calculation of gtsam.

Ok, I have turn to the GTSAM author for the reason. Maybe they can deal with the issue. As I want to use the LIO in my localization system, so the covariance is very important for snesor inforamation fusion. Do you have any material that is related to the propagation of covariance recommended for me ?

TixiaoShan, 2020-08-03:
I would suggest `Factor Graphs for Robot Perception`.

Gaochao-hit, 2020-08-04:
> I would suggest `Factor Graphs for Robot Perception`.

Thank you

closed 2020-08-04 by Gaochao-hit

Gaochao-hit, 2020-08-11:
> @Gaochao-hit
> I did notice that too. I don't know the reason behind it, probably something to do with the internal calculation of gtsam.
I have received answer about the covariance problem form the authors of GTSAM, the answer is [here](https://github.com/borglab/gtsam/issues/456). In my opinion, there seems only constraint between sequencial poses of odometry in the mapOptmization.cpp, so the covariance of the pose should be non-decreasing. Do you agree with me? And can you have a look at that question and give some feedback to them?

Gaochao-hit, 2020-08-11:
> I would suggest `Factor Graphs for Robot Perception`.

I have received answer about the covariance problem form the authors of GTSAM, the answer is [here:The mechanism in the propagation of covariance.](https://github.com/borglab/gtsam/issues/456) In my opinion, there seems only constraint between sequencial poses of odometry in the mapOptmization.cpp, so the covariance of the pose should be non-decreasing. Do you agree with me? And can you have a look at that question and give some feedback to them?

Cristian-wp, 2022-06-08:
@Gaochao-hit did you find any answer?

soldierofhell, 2022-06-15:
Hi @TixiaoShan,
I've got a question about priorNoise variance assumptions for position ([here](https://github.com/TixiaoShan/LIO-SAM/blob/4fc42fabd94da4c1b7778291627a07f7d841aa98/src/mapOptmization.cpp#L1385)). The variance is very high (1e8) and, at least in my case, it stays high despite real measurments have lower variance ([here](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/mapOptmization.cpp#L1389)). What is the rationale behind this high variance assumption. I noticed that this high variance affects acceleration bias estimation in IMUPreintegration (in my case acceleration bias is close to zero, and in reality definitely it's not). Also both /odometry/imu and /odometry/imu_incremental are very jerky, so it looks like IMU preintegration doesn't do the job properly. Thanks,

supersupercoconut, 2024-05-28:
> Hi @TixiaoShan, I've got a question about priorNoise variance assumptions for position [...]

I encountered the same problem