TixiaoShan/LIO-SAM #50: Initial rotation between GPS odometry and Optimized path

Pallav1299, 2020-08-03:
I tried the package on **Campus dataset(small)**. 

When not using GPS, the package performs very well. 
Some results:
[image]
[image]

On comparing the GPS odometry with the optimized path, both look quite similar. The initial rotation seems to be the only difference between the two.

But when using GPS, there are some serious artifacts in the output PCD. 
Some results:
[image]

Now I have the following questions:
1. What causes the initial rotation misalignment between GPS odometry and Optimized path?
2. Can we create correct(with least artifacts) global PCD map with LIO-SAM? 
3. Any method for evaluating LIO-SAM? (I tried evaluating with KITTI ground truth sequences, the elevation data was inconsistent: [LINK)](https://github.com/TixiaoShan/LIO-SAM/issues/35#issuecomment-664587989. But this can be misleading since we don't have correct IMU parameters for KITTI.)
4. How well does LIO-SAM support car-like vehicles?

TixiaoShan, 2020-08-03:
The EM field of lidar causes the yaw readings from IMU very unreliable. You can evaluate LIO-SAM using the evaluation method of KITTI, which calculates the drift at certain trajectory length. Or, you can evaluate the drift when returning to the same location for your dataset.

TixiaoShan, 2020-08-03:
How well does LIO-SAM support car-like vehicles?
I tested LIO-SAM on handheld-device, ground vehicles, and boats. UAV is not tested yet.

HaowenLai, 2020-08-06:
I have encountered the same situation as @Pallav1299 . When I tested the [park dataset](https://drive.google.com/file/d/19PZieaJaVkXDs2ZromaHTxYoq0zkiHae/view?usp=sharing) with GPS, wrong result was got. The following figure shows the GPS trajectory (the long coordinate) and Imu odom trajectory (the thin blue one, not fused with gps):

[image]

However, when I tested my self-recorded dataset with GPS, result seemed pretty good. Rotation misalignment was corrected automatically in the first few seconds. The following figure shows the GPS trajectory (the long coordinate) and Imu odom trajectory (the small one, not fused with gps):

[image]

It's quite strange that one can work but another cannot with similar GPS and IMU (not fused with gps) trajectories.

Pallav1299, 2020-08-06:
> I have encountered the same situation as @Pallav1299 . When I tested the [park dataset](https://drive.google.com/file/d/19PZieaJaVkXDs2ZromaHTxYoq0zkiHae/view?usp=sharing) with GPS, wrong result was [...]

@HaowenLai I think it is the desired result. What's happening here is that you have set the **useImuHeadingInitialization** flag to false. Since you are not using GPS, this is actually not required. Hence the result is correct.

> However, when I tested my self-recorded dataset with GPS, result seemed pretty good. Rotation misalignment was corrected automatically in the first few seconds. The following figure shows the GPS trajectory (the long coordinate) and Imu odom trajectory (the small one, not fused with gps):

The GPS factors are actual GPS poses (x,y,z), converted to local map coordinates. The optimizer tries to converge to these poses upon receiving the GPS factors. Did you set the **useImuHeadingInitialization** to **true** in this case?

If you set **useImuHeadingInitialization** to **true** with proper gpsTopic, the result will be different.

HaowenLai, 2020-08-06:
@Pallav1299 Thanks for the reply. I set **useImuHeadingInitialization** to **true** and test again the park dataset. The rotation misalignment disappears, and the trajectory looks like (GPS trajectory is with the long coordinate and Imu odom trajectory is the thin blue one, not fused with gps):

[image]

The result is acceptable. It seems that I forgot to set this parameter before. But strange thing still happens when I add GPS with mapping. As shown below, GPS and IMU trajectories match quite well in the beginning. But they diverge at a point and the mapping goes wrong. It is confusing.

[image]

yuanguobin, 2020-08-18:
I think, In essence, there is still the problem of the coordinate system not being aligned, the local Cartesian coordinate(GPS) and the body origin coordinate. Taking into account the accuracy of imu, it is not well aligned the yaw between them.

closed 2020-09-02 by stale[bot]