TixiaoShan/LIO-SAM #51: Why use two factor graph?

xieqi1, 2020-08-04:
Hi !
  I have read through your paper and code, and I find there are two factor graph in the project, so several questions have arisen:
  Q1. the result of imu preintegration seems only used to provide a initial estimate for map optimization, even not be used to deskew pointcloud when sensor move slowly, right?
  Q2. why not combine imu preintegration factor with the factors in map optimization, just like what VINS-MONO does?(use all the constraints to joint optimization simultaneously)
 
and I also have the question about Fig 1 in your paper:
  Q3. in the figure, we see the robot state node is associated with keyframe, and imu factor is provide a constraint between two successive nodes, however I find in the code, imu factor is added when receiving the pose of lidar frame(rather than keyframe), so does this figure not well match the code, or I don't understand correctly?

 Looking forward to your reply !
 Thank you !

TixiaoShan, 2020-08-05:
Q1. Right
Q2. I removed those constraints for faster optimization. 
Q3. Your statement is correct.

xieqi1, 2020-08-05:
Thank you for your answer.  I want to have a further discussion about Q2 ~
If you joint optimize imu preintegration factor can the system run in real time? and the how accuarcy of the system will improve?

TixiaoShan, 2020-08-05:
It hurts the performance a bit but can run in real-time. However, I didn't test that configuration on a large-scale dataset, which will produce a large graph.

The accuracy heavily depends on the parameters - all the noise settings. Depend on your settings, the optimization may trust IMU pre-integration or lidar odometry more. So the system becomes quite sensitive. By separating them into two different files, the system becomes much more stable.

xieqi1, 2020-08-05:
Sounds cool ! From my point of view ,this "loose coupling" system may be more robust and reliable. 
And compared with LOAM or LEGO-LOAM, a good initial estimate of odometry provided by imu preintegration make the map optimization more accuracy. So is the intital estimate play very important role in this system base on LOAM?

TixiaoShan, 2020-08-05:
Yes, initial guess is the key to scan-matching.

closed 2020-08-06 by xieqi1

YangSiri, 2020-09-14:
So, @TixiaoShan 

what exactly do _The nosie settings_  rely on? Any experience or advice on this? Thank you very much!

TixiaoShan, 2020-09-14:
@YangSiri 
Depending on the IMU, I hand-tuned those parameters to get the best performance. If the parameter is very off, you may see the robot path becomes choppy.

YangSiri, 2020-09-16:
Thank you, @TixiaoShan 

I guess it is a tricky problem like in DL. I notice you mention that the initial guess provided for scan-to-map matching is very important. Won't the iterative optimization slowly converge to a satisfied precision, which relies not that much on the initial value? Will they finally get diverged result?

TixiaoShan, 2020-09-16:
```Won't the iterative optimization slowly converge to a satisfied precision, which relies not that much on the initial value? Will they finally get diverged result? ```
Imagine you give an initial guess that is completely upside down, optimization will converge to a local optimal pose rather than a global one.
```And imu data will provide a better initial than consecutive odometry used before, right?```
If the IMU is very good with low bias and noise, the initial should be similar or better than the consecutive odometry. The consecutive odometry is still an optimization process. So its performance is effected by initial guess, number of itertation, and number of features.

YangSiri, 2020-09-17:
Yes, I get that. Thank you for your generous reply.

ZuoJiaxing, 2021-05-06:
Thanks for your detailed description. It looks like the covariances of the IMU factors and scan matching are very important in the factor graph optimization.

G4419, 2024-10-23:
About first question, why can't we use the pre-integrator to correct the point cloud? Is it because of cumulative errors?