TixiaoShan/LIO-SAM #52: How to align map with Google Earth

Tompson11, 2020-08-04:
Hi:
Thank you for sharing your work!
I've noticed that in your paper you align the map with Google Earth like the following figure, could you tell me how to do that?
[image]

TixiaoShan, 2020-08-04:
@Tompson11 
I used PowerPoint for that.

Tompson11, 2020-08-04:
@TixiaoShan 
Thank you for your reply! So It's just the alignment of 2D images instead of 3D models?

TixiaoShan, 2020-08-04:
Yes.

Tompson11, 2020-08-04:
Got it, thank you!

closed 2020-08-04 by Tompson11

zhanglei1997, 2023-02-07:
@Tompson11 
Hello. May I ask if you have successfully achieved the effect of this map display? If it is convenient, I hope you can share your specific operation. Thank you so much!

Tompson11, 2023-02-14:
> @Tompson11 Hello. May I ask if you have successfully achieved the effect of this map display? If it is convenient, I hope you can share your specific operation. Thank you so much!

Sorry, I haven't tried yet. 😂

grischi, 2023-02-15:
have you tried [rviz_satellite](https://github.com/nobleo/rviz_satellite)?