TixiaoShan/LIO-SAM #56: How to set initial start point?

Leedk3, 2020-08-05:
Hi Tixiao, and thank you for sharing your work!

I am suffering to set the initial start point. 
I want to set the starting point, not at the origin(0,0,0) but at the other place.

Could you give me any solutions?

Thanks, again.

TixiaoShan, 2020-08-05:
[You can change ```transformTobeMapped[r, p, y, x, y, z]``` here at initialization:](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/mapOptmization.cpp)
```
        // initialization
        if (cloudKeyPoses3D->points.empty())
        {
            transformTobeMapped[0] = cloudInfo.imuRollInit;
            transformTobeMapped[1] = cloudInfo.imuPitchInit;
            transformTobeMapped[2] = cloudInfo.imuYawInit;

            if (!useImuHeadingInitialization)
                transformTobeMapped[2] = 0;

            lastImuTransformation = pcl::getTransformation(0, 0, 0, cloudInfo.imuRollInit, cloudInfo.imuPitchInit, cloudInfo.imuYawInit); // save imu before return;
            return;
        }
```

Leedk3, 2020-08-05:
Thank you so much. I resolved the problem!

closed 2020-08-05 by Leedk3

qian5683, 2020-10-18:
Hi,I change the code like this 
```
 // initialization
        if (cloudKeyPoses3D->points.empty())
        {
            transformTobeMapped[0] = cloudInfo.imuRollInit;
            transformTobeMapped[1] = cloudInfo.imuPitchInit;
            transformTobeMapped[2] = cloudInfo.imuYawInit;
            transformTobeMapped[3] = 100;
            transformTobeMapped[4] = -100;
            transformTobeMapped[5] = 10;

            if (!useImuHeadingInitialization)
                transformTobeMapped[2] = 0;

            lastImuTransformation = pcl::getTransformation(0, 0, 0, cloudInfo.imuRollInit, cloudInfo.imuPitchInit, cloudInfo.imuYawInit); // save imu before return;
            return;
        }
```

but it sometimes can't initialize well as the zero start point