TixiaoShan/LIO-SAM #58: Data collection and power supply

LeisureLei, 2020-08-07:
Hi @TixiaoShan ,I'm very interested in your work. And I want to imitate your sensor matching——Ouster and 3DM-GX5-25. I wonder how you collect data from Ouster and 3DM-GX5-25. What portable data acquisition device do you take? And how to supply power? I am looking forward to your reply!

TixiaoShan, 2020-08-07:
Hi, I am using intel NUC to collect all the data. To power the devices, I use a LiPo battery and a DC voltage converter, which converts the LiPo voltage to desired device voltages.

LeisureLei, 2020-08-07:
> Hi, I am using intel NUC to collect all the data. To power the devices, I use a LiPo battery and a DC voltage converter, which converts the LiPo voltage to desired device voltages.

Thanks !

closed 2020-08-07 by LeisureLei

Amit10311, 2022-03-30:
Hi @LeisureLei, I read that you are Ouster lidar. Is it a working LiPo battery and a DC voltage converter?
 I want to power an OS1-32 (at 24 V, 1.5A) with a portable battery pack. What's your suggestion about it?