TixiaoShan/LIO-SAM #59: Result of imu preintegration becomes wrong suddenly creashes when turning

zwmisdu, 2020-08-07:
When I apply LIO  with the local dataset , after modifing the Extrinsics I find that it can perform well when the car start  moving for a  short distance  straightly .However, when the car begin to turn left, the algorithm crashes suddenly ,just as the picture shows, the scan can not match will the map and from Rviz I can see that the imu path  will be disordered.
I have check the imu data, it acc and angular velocity seem correct but the yaw has a step change (for example from 3 to  -3), it that the reason? The process of imupreintegration just take use of acc and angular velocity.
[image]

TixiaoShan, 2020-08-07:
The yaw angle is only used for initialization. I suspect the problem is caused by wrong extrinsics. So when the vehicle turns, IMU pre-integration tells to go in one direction, but lidar odometry tells to go in the opposite direction.

zwmisdu, 2020-08-08:
> The yaw angle is only used for initialization. I suspect the problem is caused by wrong extrinsics. So when the vehicle turns, IMU pre-integration tells to go in one direction, but lidar odometry tells to go in the opposite direction.

okay, I  will check the extrinsics, thank you.

closed 2020-08-08 by zwmisdu