TixiaoShan/LIO-SAM #60: Other models of IMU

WFXUWK, 2020-08-09:
Thank you for your contribution. I would like to ask if I use the mti series of inertial navigation systems to work, do I need to modify other than the modification?

Yeah2333, 2020-08-10:
I have test MTI710. Just need download ros driver from xsens. http://wiki.ros.org/xsens_mti_driver

TixiaoShan, 2020-08-10:
I don't know about your IMU. The definition of roll pitch and yaw of an IMU depends on the manufacturer. So, you have to verify it by collecting some data.

closed 2020-08-25 by stale[bot]

jiawenquan, 2021-01-06:
> I have test MTI710. Just need download ros driver from xsens. http://wiki.ros.org/xsens_mti_driverw
我使用 MTI 670 ，extrinsicRot 与 extrinsicRPY  矩阵配置有问题，出现飘移现象，可以告诉我你是怎么配置吗

Yeah2333, 2021-01-06:
@jiawenquan  in mti imu, rot and rpy have same axis, so you should set this two parameters same.

jiawenquan, 2021-01-19:
> mti imu中的@ jiawenquan，rot和rpy具有相同的轴，因此应将这两个参数设置为相同。

感谢！ 矩阵问题解决了，不过我这样有些xsens的硬件问题想请教，我设置的赫兹是400，但是实际的赫兹只有一百多，这个你知道是什么原因吗 ，有什么办法提升赫兹吗