TixiaoShan/LIO-SAM #61: Question about replacing 9-axis IMU with 6-axis IMU

xxiao-1, 2020-08-09:
Thank you for your great work!
I have noticed that "theoretically, an initialization procedure like VINS-Mono will enable LIO-SAM to work with a 6-axis IMU. " in README.md . But as far as I know, a 9-axis IMU can provide a more accurate yaw measurement than a 6-axis IMU.
So I want to ask if the yaw angle provided by imu is not accurate enough, what impact will it have on LIO-SAM？

TixiaoShan, 2020-08-10:
It the yaw angle is not accurate, the initial heading of LIO-SAM will be off a bit compared with the true heading in the earth frame. For example, you face east, but the yaw reading gives you -30~+30 around east.

xxiao-1, 2020-08-10:
OK，thank you for your answer :)

closed 2020-08-10 by xxiao-1