TixiaoShan/LIO-SAM #62: LIO-SAM initialization problem

Yeah2333, 2020-08-10:
@TixiaoShan  I have test LIO-SAM with mti-710，the result is wonderful. But I had a problem. When i start lio-sam with lidar and imu static，the result drift. What cause this problem？Is the external or internal parameters not accurate enough?But when i start lio-sam with motion ，the result is very good. 
[image]

TixiaoShan, 2020-08-10:
What kind of environment are you testing in?

Yeah2333, 2020-08-10:
in a office

TixiaoShan, 2020-08-10:
LIO-SAM doesn't work well in an indoor environment. Have you tried it outdoors?

rohansingh42, 2020-08-12:
Does it have any tunable parameters that can improve indoor performance?

TixiaoShan, 2020-08-16:
Not really. Most confined indoor environments are degraded for lidars with a small FOV.

ClementPinard, 2020-08-27:
had the same problem in my own office. the key problem is that in an office you only see vertical surfaces (the wall). If you rotate the lidar so that it can see ceiling (or ground) and the vertical walls, it should be much more stable. There is still some unresolved bias : my scan shakes around the right position. I suspect that my IMU has some inner bias and thus when standing still the gravity is not exactly compensated, resulting in a constant acceleration that messes up the imu odometry. Shouldn't the imu preintegration module of gtsam takes it into account ? Or are the bias in the code just for velocity ?

Yeah2333, 2020-08-30:
but i still have some question. i have test these scene using loam or a-loam，lego-loam，it wont drift，but in lio-sam，it drift dramatically，i think the imupreintegration  may have some bias，and because low point in map，scan to map can‘t lower the bias，so in the next ，the bias may be more massive，because last scan to map register a wrong scan 。 @TixiaoShan  @ClementPinard

TixiaoShan, 2020-08-30:
@ClementPinard 
I also notice that if you are doing indoor mapping, reduce the ```mappingCornerLeafSize``` and ```mappingSurfLeafSize``` helps a lot.

closed 2020-09-07 by stale[bot]

tenghooi, 2021-01-15:
> @ClementPinard
> I also notice that if you are doing indoor mapping, reduce the `mappingCornerLeafSize` and `mappingSurfLeafSize` helps a lot.

I reduce the 'mappingCornerLeafSize' and 'mappingSurfLeafSize' to 0.05 and 0.1 and find it works very well in indoor environment in Gazebo.  I made a video about it here [https://www.youtube.com/watch?v=aH4dnWnwABs](url)

I am new to SLAM. Would kindly explain what  'mappingCornerLeafSize' and 'mappingSurfLeafSize'  are?

[image]