TixiaoShan/LIO-SAM #64: Suffer from altitude drift

xieqi1, 2020-08-11:
Hi !
  I run LIO-SAM on my own dateset,whose path is a big circle and its length about 2.1km, but the drift in altitude about 80m, so it can't form loop constraint. and drift in x and y is acceptable. So any suggestions about how to eliminate the altitude drift?
Thank you !

daijiaoshou, 2020-08-11:
Is the IMU accuracy too low?

------------------

Best Regards

xieqi1, 2020-08-11:
@daijiaoshou Hi ! my imu  is 100Hz.

daijiaoshou, 2020-08-11:
How about the accurate of the measurement?

------------------

Best Regards
代凯

电话：13224307507
电子邮件：daikai@jluracing.com

xieqi1, 2020-08-11:
Hello ! @daijiaoshou what the measurement refer to ? if imu, it come from epson,which cost about 3000RMB,and I think its accuracy maybe enough for mapping. if lidar its Velodyne16.

daijiaoshou, 2020-08-11:
It's unreasonable to drift so much in the longitudinal direction, 

if there is any new dataset, I will share with you.

------------------

Best Regards
代凯

电话：13224307507
电子邮件：daikai@jluracing.com

TixiaoShan, 2020-08-11:
One possible reason I can think of is that the IMU intrinsics may not be accurate. For example, gravity param. So a lot of errors will accumulate along that direction.

xieqi1, 2020-08-11:
> It's unreasonable to drift so much in the longitudinal direction,  if there is any new dataset, I will share with you.
> […](#)

OK ! Thank you !

xieqi1, 2020-08-11:
Hi, @TixiaoShan  I think the initial guess of position from imu maybe not accuarcy, so I changed the imu part instead of fusion with wheel speed using kalman filter while not using acc from imu, however, it doesn't improve the perfomance. Then I set the initial delta altitude be 0 each frame, and unfortunately its altitude still suffer from huge drift.
So, does the drift come from lidar matching ?

TixiaoShan, 2020-08-11:
> Hi, @TixiaoShan I think the initial guess of position from imu maybe not accuarcy, so I changed the imu part instead of fusion with wheel speed using kalman filter while not using acc from imu, [...]

Yes, lidar scam-matching has drift all the time, the magnitude of this drift is small.

xieqi1, 2020-08-12:
@TixiaoShan So, I think this drift maybe unreasonable, and any other ways I can have a try ?

TixiaoShan, 2020-08-12:
> @TixiaoShan So, I think this drift maybe unreasonable, and any other ways I can have a try ?

Did you try to calibrate your IMU to get more accurate intrinsics? Those parameters may effect the results too.

xieqi1, 2020-08-12:
@TixiaoShan Sure, I agree accuracy of intrinsics is important, and some methods to calibrate  the intrinsics between lidar and imu?I don't find the valid ways to calibrate.
Thank you !

TixiaoShan, 2020-08-12:
[You can use this package to get the intrinsics of IMU.](https://github.com/gaowenliang/imu_utils)

xieqi1, 2020-08-13:
@TixiaoShan Thank you ! Maybe I have found the reason that cause the unreasonable result. The extrinsic parameters between lidar and baselink maybe wrong, and I will try to calibrate this param.
Thanks for your time !

closed 2020-08-14 by xieqi1