TixiaoShan/LIO-SAM #66: Dataset server has download limits

Seanmatthews, 2020-08-11:
Consider an additional or alternate server location, with unrestricted downloads, for the sample datasets. I am not able to download them due to the file being "downloaded too many times".

Seanmatthews, 2020-08-11:
Also, on at least the Park dataset, it seems I do not have the correct access to download the file.

TixiaoShan, 2020-08-11:
Do you have any suggestions? Google drive space is the only subscription I have.

Seanmatthews, 2020-08-11:
Unfortunately, I don't have any great solutions. Mega.nz offers 50GB of free storage, but either file size or download size might be limited to 5GB. Amazon AWS offers free tier S3 storage of 5GB with an allowance of 20000 GET requests. Either one of these may wok well for the smaller datasets at least.

TixiaoShan, 2020-08-12:
@Seanmatthews 
I have added a few Dropbox download links for some of the samples. The rest of the bags that are not added because they are too large.

Seanmatthews, 2020-08-14:
I think it's reasonable to ensure that at least one dataset, which works with default settings, is available. Thank you.

closed 2020-08-14 by Seanmatthews