TixiaoShan/LIO-SAM #67: unknown issue

YushengWHU, 2020-08-12:
when I successfully compile it with pcl 1.7 and gtsam 4.0.2, everything seems fine , but the launch terminal gives me
ERROR: cannot launch node of type [robot_localization/ekf_localization_node]: robot_localization
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/yushenng/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share
ERROR: cannot launch node of type [robot_localization/navsat_transform_node]: robot_localization
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/yushenng/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share

However everything seems fine while playing pre-recorded bag.
Any suggestions?

TixiaoShan, 2020-08-12:
Did you install ```robot_localization```?

YushengWHU, 2020-08-14:
Thanks,fixed

closed 2020-08-14 by YushengWHU