TixiaoShan/LIO-SAM #74: Help with ground_constraint

hitbg-yjm, 2020-08-17:
I want to add ground_constraint like lego_loam,I extracted the ground(in surfPointsFlat),and get a lidar_odom. Than I passed the lidar_odom to mapOptmization.cpp ,final I transform the lidar_odom increment to map and add the increment to pose in z(transformTobeMapped[5]).The  increment transformed to map in z is close to zero.But in my datasets ,  ground_constraint can't perform well like lego_loam. Do I miss some step ? Or is my direction wrong?  Do you have some suggestions for this?

TixiaoShan, 2020-08-18:
Your approach sounds about right. What's the problem with your implementation? Is it not accurate?

hitbg-yjm, 2020-08-20:
Maybe,I will check the code. Thanks for your response. 
A related question，three points will affect the results:
1.init guess
2.input clouds
3.Optimization
Ground_constraint in lego_loam belongs to 1 ,right?

closed 2020-08-27 by hitbg-yjm