TixiaoShan/LIO-SAM #76: Unrestricted jumps in pose when scan-matching fails

Pallav1299, 2020-08-18:
I witnessed unrestricted jumps in the optimized trajectory when scan-matching fails at some point. The poseCovariance doesn't grow with jumps, while the imuPreintegration gets reset and the GPS factors are not added. Is this the expected behavior? 
[image]

Can someone please help me with the following questions:
1. Shouldn't the jumps be restricted using GPS?
2. Shouldn't the poseCovariance increase when scan-matching fails and leads to jumps in the estimated poses? How is the poseCovariance actually calculated in GTSAM?
3. Can LIO-SAM be used in highway like environments which have homogeneous features? How robust is the GPS optimization in LIO-SAM for highway like environments?

chxyfishge, 2020-08-24:
[image]
Same as you!
When I put the lidar+imu where they are and do not move, trajectory and point cloud should stay still, but actually it slowly rotating and shaking, finally imuPreintegration gets reset.

chxyfishge, 2020-08-24:
this issue maybe same as #62

TixiaoShan, 2020-08-25:
That's very strange. What kind of hardware are you using?

TixiaoShan, 2020-08-25:
> That's very strange. What kind of hardware are you using?

Also, could you attach a picture that shows a single scan where LIO-SAM starts to drift? I want to have a better understanding of the scene.

closed 2020-09-04 by stale[bot]