TixiaoShan/LIO-SAM #77: The parameter correctionNoise

wwtinwhu, 2020-08-21:
In imuPreintegration.cpp，the lidar odometry pose noise is set by this variable. My question is how to set this parameter？ Why you set 1m? Is it too large? Thank you.
`correctionNoise = gtsam::noiseModel::Isotropic::Sigma(6, 1);`

TixiaoShan, 2020-08-25:
That parameter is manually tuned. If the timestamps of IMU and lidar are synced well, this param value can be lowered. You may try different values to find the best one for your own datasets.

wwtinwhu, 2020-08-30:
> That parameter is manually tuned. If the timestamps of IMU and lidar are synced well, this param value can be lowered. You may try different values to find the best one for your own datasets.

Thank you very much!

closed 2020-08-30 by wwtinwhu