TixiaoShan/LIO-SAM #78: How to  use LIO-SAM for solid state lidar? Such as Leishen CH32.

bage131, 2020-08-21:
Does LIO-SAM only be used on mechanical lidars? There is a very serious drift,and sometimes  it goes around in circles, when I used LIO-SAM for solid state lidar rosbag.Could you give me some advice?Thank you.

TixiaoShan, 2020-08-25:
Currently, LIO-SAM doesn't support solid-state lidar. You may need to modify "imageProjection.cpp" and "featureExtraction.cpp" to make it compatible with your hardware. I think a good start for this would be [LIVOX-LOAM](https://github.com/hku-mars/loam_livox).

bage131, 2020-08-28:
> Currently, LIO-SAM doesn't support solid-state lidar. You may need to modify "imageProjection.cpp" and "featureExtraction.cpp" to make it compatible with your hardware. I think a good start for this would be [LIVOX-LOAM](https://github.com/hku-mars/loam_livox).
-------------------------------------------------------
Thank you very much.

closed 2020-09-05 by stale[bot]