TixiaoShan/LIO-SAM #80: imageProjection and mapOptimization node do not work.

HiSeun, 2020-08-21:
Hi, I'm new to LIO-SAM.

I would like to execute the opensource with our own dataset.bag.

But... Our case, imageProjection and mapOptimization node do not work properly, so in rviz, we cannot see the saved cloud data.

every imu data, and velodyne point cloud data were published, but lio_sam/deskew/cluod_deskewed topic, and all the others with /lio_sam/* topic do not contain anything. 

We would love to get some advice for our case. Have a nice day!

HiSeun, 2020-08-21:
Even the map frame does not appear in the tf_tree

xieqi1, 2020-08-24:
I think you shoule check whether the topic names you publish the same as the subscribed topics first.

AfterJourney00, 2020-08-24:
I encountered the similar problem. I wonder where the topic "/lio_sam/mapping/odometry" comes from? According to the logic of this algorithm, "imuPreintegration" needs this topic from "mapOptimization" which is the last node.

TixiaoShan, 2020-08-25:
If the "imageProjection.cpp" is not giving anything, probably something is wrong with your data. You can try to add some debug code in "imageProjection.cpp" to see if all the data is processed correctly.

closed 2020-09-04 by stale[bot]