TixiaoShan/LIO-SAM #81: Run error when mapOptimazation

starcosmos1225, 2020-08-24:
Hi
   When I run lio-sam, an error occurred. I locate this error in the function extractNearby()(in line 834), which appears in the following code:
for (int i = numPoses-1; i >= 0; --i)
        {
            if (timeLaserInfoCur - cloudKeyPoses6D->points[i].time < 10.0)
                **surroundingKeyPosesDS->push_back(cloudKeyPoses3D->points[i]);**
            else
                break;
        }
I don't understand why this code report an error. I have checked that the pointers is not nullptr. The push_back function of pcl::point_cloud puts this PointType in a vector, changes the width to the size of the current vector, and then sets the height to 1. Why did this operation report an running error?

TixiaoShan, 2020-08-25:
Did you install the GTSAM provided in the readme. Runtime error in mapOptimization is usually caused by GTSAM.

starcosmos1225, 2020-08-27:
Thanks,
     I will try to reinstall a new version of gtsam.

ClementPinard, 2020-08-27:
Had the same error. At first I simply installed the master branch of gtsam. The only way I could make it work was to checkout precisely version 4.0.3 (as said in this repo's README) and build it without march native, otherwise I would get errors.

closed 2020-09-05 by stale[bot]

Sonne-Zhu, 2022-05-01:
> hello，
> 
> @TixiaoShan @ClementPinard
> 
> A very surprising thing, when I use the source code to compile pcl ( version 1.8.1 ) and then install it, there will be no problems. Why?

I try this，and also have no problems，although i do not know why，but thanks!