TixiaoShan/LIO-SAM #82: LIO-SAM offline ?

ClementPinard, 2020-08-27:
Hi thanks a lot for your repo, been playing with it lately and it works great !

My first goal is to have a good pointcloud of the scene i try to scan, even if it means not being realtime (I have another algorithm to localize a camera inside an already 3D mapped area, I just need the point cloud).

For the moment I run the algorithm by replayng rosbag and everything works as if it was real time. Can we do better if we don't specifically need to be real time ? Your algorithm seems to have the best result for handheld scans, even compared with offline techniques such as LIO-Mapping.

My first (naive) technique was simply to play the rosbag slower and changing params so that more key frames are added, I guess the point cloud is denser that way. I also use loop closure more extensively by increasing the radius search. Is there something else we can do ?

I am not a ROS or Lidar specialist, i just want to figure out the best way to map a scene with a handheld Lidar-imu combo, so I am open for any suggestion !

Best,
Clément

TixiaoShan, 2020-08-27:
If you don't need to run it in real-time, you can lower or increase some parameters to add keyframe more frequently and have a denser cloud:
```
odometrySurfLeafSize - 
mappingCornerLeafSize - 
mappingSurfLeafSize - 
surroundingkeyframeAddingDistThreshold - 
surroundingkeyframeAddingAngleThreshold - 
surroundingKeyframeDensity - 
surroundingKeyframeSearchRadius + (more than 100m is not necessary)
```

All IMU related parameters like those defined in ```params.yaml``` and ```imuPreintegration.cpp``` can also be tuned specifically for your IMU to get better accuracy. (The hardcoded params in imuPreintegration.cpp are tuned for my datasets.)

zwmisdu, 2020-08-30:
> If you don't need to run it in real-time, you can lower or increase some parameters to add keyframe more frequently and have a denser cloud:
>
> ```
> odometrySurfLeafSize -
> mappingCornerLeafSize - [...]

What do you mean that '(The **hardcoded params** in imuPreintegration.cpp '？

TixiaoShan, 2020-08-30:
For example, ```correctionNoise = gtsam::noiseModel::Isotropic::Sigma(6, **1**);```

closed 2020-09-08 by stale[bot]