TixiaoShan/LIO-SAM #85: The issue about orientation in sensor_msgs::Imu

junchao1212, 2020-08-28:
Hi @TixiaoShan ,
When using my IMU sensor, I need to convert the self-defined IMU message into type of "sensor_msgs::Imu" 
While I am a little confused of generating "geometry_msgs/Quaternion orientation", what is the frame of it?  
The relevant infomation in my message is "heading","pitch","roll"(rad),where "heading" is the angle between IMU-X and due North. Coule I directly use  tf::createQuaternionMsgFromRollPitchYaw(roll, pitch, heading) or  do a relative transform to the first frame data? Then how to calibrate extQRPY?

Hope to get some suggestions from you, thank you!

TixiaoShan, 2020-08-28:
extQRPY transforms roll, pitch, and yaw measurements from IMU frame to REP-105 convention. If you are using a self-defined IMU message, you can set extQRPY to be an identity matrix, then make sure your IMU roll, pitch, and yaw follow the REP-105 convention. So ```tf::createQuaternionMsgFromRollPitchYaw(roll, pitch, heading)``` is good to do the job.

junchao1212, 2020-08-29:
Ok，thanks！

closed 2020-08-29 by junchao1212