TixiaoShan/LIO-SAM #86: there is not gpsfactor when I test Park.bag

Maxinjun, 2020-08-28:
When I test Park.bag, I found there is  not gpsfactor in the graph, gpsQueue.empty() is true.
So I check the code,
(1) in params.xml.  gpsTopic is "odometry/gpsz"  it should be "odometry/gps" 
(2) in module_navsat.launch, navsat_transform_node will publish  "odometry/gps" , but there is not input topic "odometry/navsat".
AnyOne know how to set params correctly?

TixiaoShan, 2020-09-01:
You can also increase ```gpsCovThreshold``` and decrease ```poseCovThreshold``` to have more GPS factors inserted.

Maxinjun, 2020-09-02:
I add some logs in mapOptmization.cpp  
(1) void gpsHandler(const nav_msgs::Odometry::ConstPtr& gpsMsg)
    {
		ROS_INFO("receive gspMsg");
        gpsQueue.push_back(*gpsMsg);
    }

(2)  void addGPSFactor()
    {
		ROS_INFO("addGPSFactor");
        if (gpsQueue.empty())
		{
			ROS_INFO("gpsQueue.empty()");
            return;
		}
		
        // wait for system initialized and settles down
        if (cloudKeyPoses3D->points.empty())
            return;
        else
        {
            if (pointDistance(cloudKeyPoses3D->front(), cloudKeyPoses3D->back()) < 5.0)
                return;
        }

        // pose covariance small, no need to correct
        if (poseCovariance(3,3) < poseCovThreshold && poseCovariance(4,4) < poseCovThreshold)
            return;
here is the log
[ INFO] [1599018304.927228063]: gpsQueue.empty()
[ INFO] [1599018305.215031066]: addGPSFactor
[ INFO] [1599018305.215055680]: gpsQueue.empty()
[ INFO] [1599018305.477979032]: addGPSFactor
[ INFO] [1599018305.478002703]: gpsQueue.empty()
[ INFO] [1599018305.754783888]: addGPSFactor
[ INFO] [1599018305.754805628]: gpsQueue.empty()
[ INFO] [1599018306.020387800]: addGPSFactor
[ INFO] [1599018306.020414215]: gpsQueue.empty()
[ INFO] [1599018306.308104136]: addGPSFactor
[ INFO] [1599018306.308124551]: gpsQueue.empty()
[ INFO] [1599018306.576110485]: addGPSFactor
[ INFO] [1599018306.576137974]: gpsQueue.empty()
[ INFO] [1599018306.833912850]: addGPSFactor
[ INFO] [1599018306.833945892]: gpsQueue.empty()
[ INFO] [1599018307.140664706]: addGPSFactor
[ INFO] [1599018307.140695145]: gpsQueue.empty()
[ INFO] [1599018307.406439835]: addGPSFactor
[ INFO] [1599018307.406461485]: gpsQueue.empty()
[ INFO] [1599018307.681494864]: addGPSFactor
[ INFO] [1599018307.681520723]: gpsQueue.empty()
[ INFO] [1599018307.934242893]: addGPSFactor
[ INFO] [1599018307.934271740]: gpsQueue.empty()
[ INFO] [1599018308.292291682]: addGPSFactor
[ INFO] [1599018308.292313551]: gpsQueue.empty()
[ INFO] [1599018308.587932193]: addGPSFactor
[ INFO] [1599018308.587955497]: gpsQueue.empty()
[ INFO] [1599018308.858106626]: addGPSFactor
[ INFO] [1599018308.858136649]: gpsQueue.empty()
[ INFO] [1599018309.146306650]: addGPSFactor
[ INFO] [1599018309.146326503]: gpsQueue.empty()
[ INFO] [1599018309.394690033]: addGPSFactor
[ INFO] [1599018309.394709316]: gpsQueue.empty()

I check the param.yaml  
  gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file
 It should be  "odometry/gps" after I check http://docs.ros.org/melodic/api/robot_localization/html/navsat_transform_node.html
Subscribed Topics is imu/data, odometry/filtered, gps/fix 
in module_navsat.launch
      <remap from="imu/data" to="imu_correct" />
        <remap from="gps/fix" to="gps/fix" />
        <remap from="odometry/filtered" to="odometry/navsat" />
which module publish odometry/navsat?  because lack of this topic , there is not gps msg as I think

Maxinjun, 2020-09-03:
it is my issue, I don't install robot-localization package

closed 2020-09-03 by Maxinjun

HaisenbergPeng, 2021-04-27:
@Maxinjun Hi, did you get odometry/gps published in the end?

yinhaosen, 2023-03-27:
Did you manage to see the vehicle back to the start point? I tried many times it always ended up with a position long away from the start point.

wwtinwhu, 2023-04-03:
> Did you manage to see the vehicle back to the start point? I tried many times it always ended up with a position long away from the start point.

Hello, @yinhaosen . I noticed that the provided bag file is part of the Park dataset decribed in the paper. The park dataset is around 40 min. While the bag is around 9 min.

yinhaosen, 2023-04-03:
> Hello, @yinhaosen . I noticed that the provided bag file is part of the Park dataset decribed in the paper. The park dataset is around 40 min. While the bag is around 9 min.

Exactly. I guess this bag is just a small sample not the real one tested in the paper. But I still can use it to compare slam poses with gps poses as ground truth in order to replicate the paper results (e.g RMSE) using EVO ape.

wwtinwhu, 2023-04-03:
> Exactly. I guess this bag is just a small sample not the real one tested in the paper. But I still can use it to compare slam poses with gps poses as ground truth in order to replicate the paper [...]

Hi,@yinhaosen. I extracted the gps position from bag. However, the evaluation result is not right. Look for help. 
I converted the lon and lat to the x, y coordinate using https://github.com/weisongwen/GraphGNSSLib/blob/main/global_fusion/include/gnss_tools.h#L220. I added the align option when comparing use evo.
[image]

yinhaosen, 2023-04-03:
I just recorded two topics /odometry/gps and /lio_sam/mapping/odometry to compare.

wwtinwhu, 2023-04-03:
@yinhaosen Thanks! My way have also woked just now. 
[image]