TixiaoShan/LIO-SAM #87: Sparkfun SEN-14001 IMU (MPU2950 chip)

GilenTell, 2020-08-28:
Hi,
I am trying to integrate the SEN-14001 IMU. I have calibrated the IMU and I have used the ROS package driver from https://github.com/KristofRobot/razor_imu_9dof.

I am launching lio_sam with real-time data of this IMU and velodyne VLP16 Lidar (in my case, the compatible LeiShen).
But the mapping is not stable. Even though the equipment is in a static position, the drawing of the instant map of point cloud is scrolling across the RViz screen and moving quickly and erratically.

Aren't this IMU component characteristics of noise and precision enough for LIO-SAM algorithms?
Is there any params.yaml configuration that I could change to use SEN-14001?
Has anybody used this IMU with LIO-SAM?
Any recommendation of other IMU cheaper than Microstrain to integrate with LiO-SAM?

Thank you and regards,
Gilen

TixiaoShan, 2020-08-28:
I think your IMU is good enough to run the package. Just make sure the extrinsics for IMU acceleration and orientation are correct. The provided extrinsics in the package may not be suitable for your IMU because the IMU orientation definition might be different depends on the manufacturer.

rostesting, 2020-08-29:
Regarding alternative IMUs, we're using a PhidgetSpatial Precision 3/3/3 with the `phidgets_imu` and `imu_filter_madgwick` packages. It's about $120, and they have a less expensive version with lower sensitivity. LiDAR is an OS1-64. However, I haven't had any success getting it to work with LIO-SAM. I'm not sure what I'm doing wrong ... but I'll leave that for another post.

GilenTell, 2020-08-31:
Thank you for your quick response!
I think that I have configured the extrinsic right. These are my matrix config:
```
  # Extrinsics (lidar -> IMU)
  extrinsicTrans: [0.0, 0.0, 0.0]
  extrinsicRot: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]
  extrinsicRPY: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]
```
And I have validated the RPY rolling directions positive and negative compared to your image and they are OK.
[image]

Below is a log of LIO-SAM IMU debug data. I suspect that IMU acc noise is too high. Isn't it?

Here is an image of my testbed with IMU mounted on top of LiDAR.
[image]

A capture of RViz with Mapping local visualization to apreciate the displacement. This is the point cloud of the room of our office.
[image]

Any advice or guidance?
Thank you and regards,
Gilen

```
IMU acc: 
x: -0.24515, y: 0.160114, z: 9.67155
IMU gyro: 
x: 0, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00401426, pitch: 0.0221657, yaw: 0.12898

IMU acc: 
x: -0.207611, y: 0.155134, z: 9.66695
IMU gyro: 
x: -0.01, y: 0.01, z: -0.01
IMU roll pitch yaw: 
roll: 0.00506145, pitch: 0.0226893, yaw: 0.128805

IMU acc: 
x: -0.0375386, y: 0.103423, z: 9.74586
IMU gyro: 
x: -0.01, y: 0.01, z: -0.01
IMU roll pitch yaw: 
roll: 0.00593412, pitch: 0.0225147, yaw: 0.128805

IMU acc: 
x: -0.24515, y: -0.0800568, z: 9.5371
IMU gyro: 
x: 0, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00645772, pitch: 0.0207694, yaw: 0.129154

IMU acc: 
x: -0.155517, y: 0.24017, z: 9.39806
IMU gyro: 
x: 0, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00523599, pitch: 0.0211185, yaw: 0.129503

IMU acc: 
x: -0.0850364, y: 0.221018, z: 9.62061
IMU gyro: 
x: 0, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00715585, pitch: 0.0207694, yaw: 0.130376

IMU acc: 
x: -0.217188, y: 0.0800568, z: 9.64359
IMU gyro: 
x: -0.01, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00855211, pitch: 0.0195477, yaw: 0.129503
IMU acc: 
x: -0.24515, y: 0.160114, z: 9.67155
IMU gyro: 
x: 0, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00401426, pitch: 0.0221657, yaw: 0.12898

IMU acc: 
x: -0.207611, y: 0.155134, z: 9.66695
IMU gyro: 
x: -0.01, y: 0.01, z: -0.01
IMU roll pitch yaw: 
roll: 0.00506145, pitch: 0.0226893, yaw: 0.128805

IMU acc: 
x: -0.0375386, y: 0.103423, z: 9.74586
IMU gyro: 
x: -0.01, y: 0.01, z: -0.01
IMU roll pitch yaw: 
roll: 0.00593412, pitch: 0.0225147, yaw: 0.128805

IMU acc: 
x: -0.24515, y: -0.0800568, z: 9.5371
IMU gyro: 
x: 0, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00645772, pitch: 0.0207694, yaw: 0.129154

IMU acc: 
x: -0.155517, y: 0.24017, z: 9.39806
IMU gyro: 
x: 0, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00523599, pitch: 0.0211185, yaw: 0.129503

IMU acc: 
x: -0.0850364, y: 0.221018, z: 9.62061
IMU gyro: 
x: 0, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00715585, pitch: 0.0207694, yaw: 0.130376

IMU acc: 
x: -0.217188, y: 0.0800568, z: 9.64359
IMU gyro: 
x: -0.01, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00855211, pitch: 0.0195477, yaw: 0.129503

IMU acc: 
x: 0.0283455, y: 0.0279624, z: 9.69951
IMU acc: 
IMU gyro: 
x: -0.01, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00837758, pitch: 0.0198968, yaw: 0.128107

IMU acc: 
x: -0.25013, y: 0.0609045, z: 9.32834
IMU gyro: 
x: -0.01, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00785398, pitch: 0.0176278, yaw: 0.12898

IMU acc: 
x: -0.25013, y: 0.122192, z: 9.48156
IMU gyro: 
x: 0, y: 0, z: 0
IMU roll pitch yaw: 
roll: 0.00785398, pitch: 0.018326, yaw: 0.130202

IMU acc: 
x: 0.0283455, y: 0.0279624, z: 9.69951
IMU gyro: 
x: -0.01, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00837758, pitch: 0.0198968, yaw: 0.128107

IMU acc: 
x: -0.25013, y: 0.0609045, z: 9.32834
IMU gyro: 
x: -0.01, y: 0, z: -0.01
IMU roll pitch yaw: 
roll: 0.00785398, pitch: 0.0176278, yaw: 0.12898

IMU acc: 
x: -0.25013, y: 0.122192, z: 9.48156
IMU gyro: 
x: 0, y: 0, z: 0
IMU roll pitch yaw: 
roll: 0.00785398, pitch: 0.018326, yaw: 0.130202
```

GilenTell, 2020-08-31:
I have noticed that with LeGO-LOAM packet, there is a drift too.
I suspect that the cause could be that my test enviroment is indoor.
It seems that is a well-known issue with LeGO-LOAM.
https://github.com/RobustFieldAutonomyLab/LeGO-LOAM/issues/181

Is there any configuration that optimize indoor locationn performance?

Thanks,
Gilen

GilenTell, 2020-09-01:
Hi again!
I have encountered the cause of the issue. It is the LeiShen LiDAR.
I have changed it, and I have done the same test with a Velodyne Puck Lite, and the odemetry is running perfect and there isn't any drift.
Thank you and I'm sorry if this issue has lost you time.
Gilen

closed 2020-09-01 by GilenTell

NicoNorena, 2020-09-04:
@GilenTell  Did you run this test indoors? I am running it indoors with the Puck Lite but the odometry  is drifting all over .

HappySamuel, 2020-11-02:
Hi @GilenTell 

I saw that you're using the leishen lidar with the LIO-SAM. I am currently use the same lidar. Please provide some hints/guidance on how to make the LIO-SAM take in the pointcloud from leishen lidar. As from what i found, leishen lidar's pointcloud only consist of X, Y, Z, intensity. There's neither ring nor time are exist in the data.

Best,
Samuel

GilenTell, 2020-11-03:
Hi,
The Leishen LiDAR's protocolo in its default mode is compatible with Velodyne's with same UDP packet format.
I have implemented de LOAM algorithms in ROS with velodyne driver:
http://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20Velodyne%20VLP16

Nevertheless, I encountered some problems with Leishen and its quality for mapping and now I am working with a Velodyne PUCK Lite.

Best,
Gilen

HappySamuel, 2021-06-22:
@GilenTell 

I have tried to edit the imageProjection.cpp to be compatible with Leishen lidar, below are screenshots of what i added:

[image]

[image]

[image]

[image]

And also edit in utility.h for below:

[image]

[image]

I manage to get the LIO-SAM to subscribe to Leishen Lidar without altering its pointcloud. But it will drift away, no matter i use the  cheap SEN-14001 IMU or expensive microstrain-3DM-GX5.

@TixiaoShan 
Do you know why? and is there alternative solution for me to get Leishen Lidar to work with your LIO-SAM? because the map that your LIO-SAM built is super good.

Best,
Samuel