TixiaoShan/LIO-SAM #90: runtime synchronization error

YushengWHU, 2020-08-30:
I recorded an ouster os1-64 bag with external IMU, and everything works fine for Lego_LOAM, but when I turn to LIO-SAM, nothing showed in rviz, thus I checked the code and found the synchronization error below.
The first figure is from myself recored bag, which indicates the time is the internal ouster time, while the second is from your dataset, where both timestamp is ros time.
[image]
[image]

TixiaoShan, 2020-08-30:
There is about 50 years time difference between your IMU and Ouster data. Have you tried to sync your Ouster lidar time with the current time?

zwmisdu, 2020-08-30:
> I recorded an ouster os1-64 bag with external IMU, and everything works fine for Lego_LOAM, but when I turn to LIO-SAM, nothing showed in rviz, thus I checked the code and found the synchronization [...]

I have the same problem and handled.But I could not download the ouster dataset because the wall.Are you from China too?

YushengWHU, 2020-08-31:
> I have the same problem and handled.But I could not download the ouster dataset because the wall.Are you from China too?

留个邮箱我发给你老铁

YushengWHU, 2020-08-31:
> There is about 50 years time difference between your IMU and Ouster data. Have you tried to sync your Ouster lidar time with the current time?

We

> I have the same problem and handled.But I could not download the ouster dataset because the wall.Are you from China too?

哦对，你怎么弄啊，是录bag的时候搞，还是用sim time？

YushengWHU, 2020-08-31:
> I have the same problem and handled.But I could not download the ouster dataset because the wall.Are you from China too?

哦对，你这个问题是怎么解决的哈~也是全弄sim time还是录bag时候搞？

zwmisdu, 2020-08-31:
> 留个邮箱我发给你老铁

zwmjust@126.com 邮箱留个微信吧 交流一下

closed 2020-08-31 by YushengWHU