TixiaoShan/LIO-SAM #92: what is the difference between these topics

xxiao-1, 2020-08-31:
Hello.Thank you for your great work! I am a beginner.I have some questions about the meaning of the topics.
Cloud you please tell me the difference between odomTopic+"_incremental" and odomTopic in "imuPreintegration.cpp",and the difference between "/lio_sam/mapping/odometry" and "/lio_sam/mapping/odometry_incremental" topic in "mapOptmization.cpp"?

Thank you very much!

TixiaoShan, 2020-09-01:
```/lio_sam/mapping/odometry``` is the odometry with pose correction, thus it has no drift.
```/lio_sam/mapping/odometry_incremental``` is pure lidar odometry without any correction, thus it has drifts.

xxiao-1, 2020-09-02:
Thank you for your answer.
If **/lio_sam/mapping/odometry** provides a better pose, why we use **/lio_sam/mapping/odometry_incremental** in "imuPreintegration.cpp"?

xieqi1, 2020-09-07:
"imuPreintegration.cpp" needs a constraint between two successive frames from lidar odometry,   although  /lio_sam/mapping/odometry_incremental suffer from drift it is accurate enough for  successive frames. If use /lio_sam/mapping/odometry you will suffer from pose jump for example when detect loop closure.

closed 2020-09-15 by stale[bot]