TixiaoShan/LIO-SAM #94: IMU error?

YushengWHU, 2020-09-01:
Hello here, the IMU am using have the same coordinate definition with ouster os1-64, which is front-left-up, thus I change the "extrinsicRot" and "extrinsicRPY" to identity matrices, and debug IMU according to the video, everything's fine. However, when I play my recorded bag, the result is a disaster, which seems to be IMU error.
[image]
And the terminal gives me 
[image]
Do you have any suggestions? By the way, I've tested the same IMU with ouster using Lego-Loam, which is fine.
If you are interested, the recorded bag is 
[https://drive.google.com/file/d/10zUV7I8Mh7suLumHvFeLKOO-yTBr19wj/view?usp=sharing](url)

TixiaoShan, 2020-09-01:
It seems that the cloud from your lidar is not in REP format.

Change code in ```projectPointCloud()``` to
```
thisPoint.x = -laserCloudIn->points[i].x;
thisPoint.y = -laserCloudIn->points[i].y;
thisPoint.z = laserCloudIn->points[i].z;
```

You will notice that the IMU prediction is very bad, this is caused by bad time sync between the lidar and IMU.
[image]

YushengWHU, 2020-09-02:
Could you explain why this bad time sync happen? Cause from the terminal time debug, everthing seems fine compared with ur ouster dataset, I also tried using sim time for implementation and get bad results. I use a very simple and low-cost IMU for tests (lower than $50) with 200Hz publish frequency, should it be the reason?

TixiaoShan, 2020-09-02:
It is not an IMU problem, it's Outer. Ouster lidar sometimes gives me timestamp in the future, sometimes gives timestamp a few seconds behind the system time.

YushengWHU, 2020-09-02:
Ok, weird, as am using PTP sync, the timestamp is actually given by external sources, how could Ouster given timestamp in the future or behind the system time, hard to understand.

TixiaoShan, 2020-09-02:
```how could Ouster given timestamp in the future or behind the system time, hard to understand.```
That's also what confuses me. Ouster lidar is not stable at all.

closed 2020-09-03 by YushengWHU

dllu, 2020-09-03:
the quality of PTP time syncing depends on the ethernet hardware you're using, and whether you're using a router or switch. It can also take a few seconds (or a few minutes) to establish stable time syncing.

YushengWHU, 2020-09-04:
> the quality of PTP time syncing depends on the ethernet hardware you're using, and whether you're using a router or switch. It can also take a few seconds (or a few minutes) to establish stable time syncing.

Ok, thx, I thought I've successfully synced my lidar and imu, cause a friend of mine have generated  a beautiful loop map with my recorded bag, 
[image]
I use a DJI manifold2 to record rosbag, with a DJI router connecting different lidars to manifold.

YushengWHU, 2020-09-04:
Just a few things to notice when applying LIO-SAM with your recorded bag
1. You should sync the IMU and lidar 'precisely' either by time reference(GPS, PTP, whatever) or by changing the timestamp of both IMU and lidar to ros::Time::now(), both works fine for me.
2. Check ur coordinates definition
3. Make sure you have calibrated ur IMU and modify the param setting, especially when you are using a pretty simple one(for me, a $30 IMU)
4. If you wanna record ur GPS with IMU, and u want to omit the confusing time sync (self decoding, serial port processing), u better record ur bag like what was done in park.bag.

TixiaoShan, 2020-09-04:
> the quality of PTP time syncing depends on the ethernet hardware you're using, and whether you're using a router or switch. It can also take a few seconds (or a few minutes) to establish stable time syncing.

@dllu 
I connect Ouster to the NUC directly. I know that it takes a few minutes to establish sync when the sensor is booted. After the timestamp is synced using PTP, however, I would get -0.5 ~ +0.5 timestamp difference when running ```rostopic delay /lidar_topic```. Most of the time, the delay is around +0.1.

TixiaoShan, 2020-09-04:
@YushengWHU 
It's good to know it's working now. How much delay do you encounter in your point cloud when you run ```rostopic delay```?

YushengWHU, 2020-09-05:
> @YushengWHU
> It's good to know it's working now. How much delay do you encounter in your point cloud when you run `rostopic delay`?
Well, that's a point, I find that my /imu delay is 0s, while the /lidar_points delay is  2.15s on average

TixiaoShan, 2020-09-05:
> > @YushengWHU
> > It's good to know it's working now. How much delay do you encounter in your point cloud when you run `rostopic delay`?
> > Well, that's a point, I find that my /imu delay is 0s, while the /lidar_points delay is  2.15s on average

That's a very large delay... My average lidar delay is only around 0.125s.

YushengWHU, 2020-09-05:
> > @YushengWHU
> > It's good to know it's working now. How much delay do you encounter in your point cloud when you run `rostopic delay`?
> > Well, that's a point, I find that my /imu delay is 0s, while the /lidar_points delay is  2.1

> That's a very large delay... My average lidar delay is only around 0.125s.

Sure, and  I want to test whether the 2.15s delay is steady or not, if yes, I can keep sync 'precisely' with IMU, unfortunatelly, the delay is step by step increasing after booted, for my case, around a 0.001s increasing per minute.

Maxinjun, 2020-09-09:
@YushengWHU 
你好,我遇到了类似的问题,如你图片所示, 跑起来就飞了, 预积分也报那个警告
请问你是怎么修复的?我看到你的note了,我用的ros::Time::now()

YushengWHU, 2020-09-09:
> @YushengWHU
> 你好,我遇到了类似的问题,如你图片所示, 跑起来就飞了, 预积分也报那个警告
> 请问你是怎么修复的?我看到你的note了,我用的ros::Time::now()

你用的os1还是os2 如果是os1你要按作者说的方式改一下坐标轴，这个警告都是坐标轴的问题

Maxinjun, 2020-09-09:
@YushengWHU,
我用的是os1, 看ouster datasheet, 它又有sensor frame, lidar frame, ouster输出的点云好像又是sensor系下
能加下微信好友吗? 18021501120, 想交流下,

YushengWHU, 2020-09-10:
> @YushengWHU,
> 我用的是os1, 看ouster datasheet, 它又有sensor frame, lidar frame, ouster输出的点云好像又是sensor系下
> 能加下微信好友吗? 18021501120, 想交流下,

你看上面作者说的试试看，应该就可以了，我的没问题

Maxinjun, 2020-09-10:
@YushengWHU 
你用的IMU输出的orientation参考的世界坐标系是哪个? ENU  NED 还是初始时刻

YushengWHU, 2020-09-10:
> @YushengWHU
> 你用的IMU输出的orientation参考的世界坐标系是哪个? ENU NED 还是初始时刻

车体坐标系是前右上

aliakbarmostafaei, 2020-09-15:
@YushengWHU @dllu I've been having issues trying to make this work with an Ouster OS1, I have tried the solutions suggested in this issue, re-syncing timestamps, trying different frame coordinations in params.yaml but to no avail! The problem is after a few seconds of playing the bag, the base_link drops altitude and in it keeps decreasing in Z axis!
I've opened this issue here #111 but the author's comments didn't offer much help, could you maybe take a look and let me know what you think the culprit is?
Thank you!

bromeara, 2020-12-21:
> It seems that the cloud from your lidar is not in REP format.
>
> Change code in `projectPointCloud()` to
>
> ```
> thisPoint.x = -laserCloudIn->points[i].x;
> thisPoint.y = -laserCloudIn->points[i].y; [...]

Which REP are you referencing for the orientation of pointclouds?

andoreso, 2020-12-23:
@bromeara 
> Which REP are you referencing for the orientation of pointclouds?

I believe he is referencing to REP 103 frame conventions
[https://www.ros.org/reps/rep-0103.html#coordinate-frame-conventions](https://www.ros.org/reps/rep-0103.html#coordinate-frame-conventions)

bromeara, 2020-12-23:
@andoreso I also thought so but also I found that curious because it doesn't define a default coordinate frame for lidar sensors only for bodies. With the OS there are 3 frames published, os_sensor which is the parent of the other two frames is located at the base of the lidar with the X facing away from the cable port(Probably is normally facing the front of the vehicle but maybe its the cause of the problem), the os_lidar x-axis is then aligned with the encoder count which is facing towards the cable connector(The front of the lidar_sensor but probably not the vehicle) and then os_imu has the frames aligned to its axis. These seem to comply with 103, and I maintain a fork of the ouster driver so I was curious if there was actually a REP I should be complying with.

andoreso, 2020-12-23:
@bromeara 
I understood from the standard that all coordinate frames should be transformed to the axis conventions with:
X = forward
Y = left
Z = up
So the os_sensor frame is in the correct frame but I believe that lio-sam ignores transforms and uses the raw PointCloud2 that is in the os_lidar frame. So the transform should be manually applied to the point cloud. I think the easy solution is to subscribe to the point cloud invert the xy values and republish it, but there may be a better solution.

Also please take what I said with a grain of salt because I just started testing lio-sam with a lidar but the inversion of x and y has worked for me in other lidar projects using ouster lidar.

hummblebee, 2021-03-01:
Ouster has released a new firmware (v2.0), is LIO-SAM compatible with the new changes mentioned [here](https://data.ouster.io/downloads/changelog/changelog-v2p0.pdf)?

bromeara, 2021-03-03:
> Ouster has released a new firmware (v2.0), is LIO-SAM compatible with the new changes mentioned [here](https://data.ouster.io/downloads/changelog/changelog-v2p0.pdf)?

We have run it on v2.0 and this is more of a question for your ouster ROS driver than this library

yxallnyy, 2021-04-28:
> Hello here, the IMU am using have the same coordinate definition with ouster os1-64, which is front-left-up, thus I change the "extrinsicRot" and "extrinsicRPY" to identity matrices, and debug IMU [...]

hi,could you share me the dataset and config . just for the test , be easy.thanks.

YushengWHU, 2021-05-04:
> hi,could you share me the dataset and config . just for the test , be easy.thanks.

leave me your email, I will send all the required to you.

nhk035, 2021-05-08:
@YushengWHU 
老哥，能说一下你用的什么型号的IMU呀，想整一个便宜的玩玩。多谢啦

YushengWHU, 2021-05-08:
> @YushengWHU
> 老哥，能说一下你用的什么型号的IMU呀，想整一个便宜的玩玩。多谢啦

我用的xsens的mti680G， 2W3，便宜的你就淘宝搜ROS IMU，整个200块的就行，我以前买过200块的，liosam也能跑

nhk035, 2021-05-08:
> > @YushengWHU
> > 老哥，能说一下你用的什么型号的IMU呀，想整一个便宜的玩玩。多谢啦
> 
> 我用的xsens的mti680G， 2W3，便宜的你就淘宝搜ROS IMU，整个200块的就行，我以前买过200块的，liosam也能跑

感谢回复

forcazzx, 2021-07-21:
> Ok, thx, I thought I've successfully synced my lidar and imu, cause a friend of mine have generated a beautiful loop map with my recorded bag,
> [image]
> I use a DJI manifold2 to record rosbag, with a [...]

请问你这是在武大信操跑的吗，我也准备采一组数据，你能把该数据和一些设置分享给我吗，我想测试一下，我的邮箱：zaixing@whu.edu.cn，谢谢～

YushengWHU, 2021-07-21:
> > > the quality of PTP time syncing depends on the ethernet hardware you're using, and whether you're using a router or switch. It can also take a few seconds (or a few minutes) to establish stable time syncing.
> > 
> > 
> > Ok, thx, I thought I've successfully synced my lidar and imu, cause a friend of mine have generated a beautiful loop map with my recorded bag,
> > [image]
> > I use a DJI manifold2 to record rosbag, with a DJI router connecting different lidars to manifold.
> 
> 请问你这是在武大信操跑的吗，我也准备采一组数据，你能把该数据和一些设置分享给我吗，我想测试一下，我的邮箱：[zaixing@whu.edu.cn](mailto:zaixing@whu.edu.cn)，谢谢～

青楼跑的，数据不小，还有一些别的数据，有空你加我微信IchliebeYe，你来我这拷一下就行，我在GPS中心

llxClover, 2021-08-28:
> Hello here, the IMU am using have the same coordinate definition with ouster os1-64, which is front-left-up, thus I change the "extrinsicRot" and "extrinsicRPY" to identity matrices, and debug IMU [...]

I have met the same trouble,could you tell me what method you used to make it?

llxClover, 2021-08-28:
> Ok, thx, I thought I've successfully synced my lidar and imu, cause a friend of mine have generated a beautiful loop map with my recorded bag,
> [image]
> I use a DJI manifold2 to record rosbag, with a [...]

Could you tell me how to sync my bag?
Thank you !

YushengWHU, 2021-08-30:
> Could you tell me how to sync my bag?
> Thank you !

你是说咋把雷达和IMU同步是吧，你用的啥雷达和IMU，有啥具体问题加我微信问，IchliebeYe

juliangaal, 2021-10-15:
> It seems that the cloud from your lidar is not in REP format.
>
> Change code in `projectPointCloud()` to
>
> ```
> thisPoint.x = -laserCloudIn->points[i].x;
> thisPoint.y = -laserCloudIn->points[i].y; [...]

Wouldn't this be configurable in the imu -> lidar portion of the config yaml file as well?

zyx-99, 2022-04-30:
> > @YushengWHU
> > 老哥，能说一下你用的什么型号的IMU呀，想整一个便宜的玩玩。多谢啦
> 
> 我用的xsens的mti680G， 2W3，便宜的你就淘宝搜ROS IMU，整个200块的就行，我以前买过200块的，liosam也能跑

想请问您一下 我是用的3dm gx5 25imu 为什么从第一秒就开始飞起来了呢

HandsomeJunJun, 2022-06-24:
老哥，你的微信加不上啊。。

zyx-99, 2022-06-24:
> 老哥，你的微信加不上啊。。

兄弟 你跑通了嘛 用的什么型号的imu？

HandsomeJunJun, 2022-06-24:
> 
和作者的差不多型号的，3dm-gx4-25,不过时间同步没有成功，也正在问作者。。。

HandsomeJunJun, 2022-06-24:
> > 老哥，你的微信加不上啊。。
> 
> 兄弟 你跑通了嘛 用的什么型号的imu？

和作者的差不多型号的，3dm-gx4-25,不过时间同步没有成功，也正在问作者。。。

zyx-99, 2022-06-24:
> > > 老哥，你的微信加不上啊。。
> > 
> > 
> > 兄弟 你跑通了嘛 用的什么型号的imu？
> 
> 和作者的差不多型号的，3dm-gx4-25,不过时间同步没有成功，也正在问作者。。。

我是3dm-gx5-25 我是xyz和欧拉角没有对上...一启动就z轴飞天 你是用的microstrain官方驱动吗

zyx-99, 2022-06-24:
> > > 老哥，你的微信加不上啊。。
> > 
> > 
> > 兄弟 你跑通了嘛 用的什么型号的imu？
> 
> 和作者的差不多型号的，3dm-gx4-25,不过时间同步没有成功，也正在问作者。。。

方便联系一下嘛

YushengWHU, 2022-06-24:
> IchliebeYe

我微信就这个啊。。IchliebeYe

HandsomeJunJun, 2022-06-27:
你加楼下老哥的微信吧，我的是时间同步没有成功。。。。

HandsomeJunJun, 2022-06-27:
> > > > 老哥，你的微信加不上啊。。
> > > 
> > > 
> > > 兄弟 你跑通了嘛 用的什么型号的imu？
> > 
> > 
> > 和作者的差不多型号的，3dm-gx4-25,不过时间同步没有成功，也正在问作者。。。
> 
> 我是3dm-gx5-25 我是xyz和欧拉角没有对上...一启动就z轴飞天 你是用的microstrain官方驱动吗
是的，官方驱动，老是有问题。。。时间同步不太成功。。。

zhuoyuan0, 2022-06-29:
> > IchliebeYe
> 
> 我微信就这个啊。。IchliebeYe

can not add you either

18713472531, 2022-08-08:
> > > > 老哥，你的微信加不上啊。。
> > > 
> > > 
> > > 兄弟 你跑通了嘛 用的什么型号的imu？
> > 
> > 
> > 和作者的差不多型号的，3dm-gx4-25,不过时间同步没有成功，也正在问作者。。。
> 
> 我是3dm-gx5-25 我是xyz和欧拉角没有对上...一启动就z轴飞天 你是用的microstrain官方驱动吗

兄弟怎么解决的，我用test your slam那篇论文里面的subt-t-tunnel数据集，一跑起来就起飞，不知道该怎么解决

JACKLiuDay, 2022-11-30:
> Hi @YushengWHU
> 
> Regarding the PTP for ouster lidar, you can actually do a little hack on the ouster_ros package to make it to use ros::Time::now() when sending out /ouster_cloud_node/imu and /ouster_cloud_node/points, then the future or outdated timestamp issue will be ruled out.
> 
> Best, Samuel

Hi, I saw your reply about changing ouster_ros driver to make sync lidar point to ros::Time::now(). Could you give some help. I don't know where to change the code. Thank you so much.

juliangaal, 2022-11-30:
This is the line you're looking for:

https://github.com/ouster-lidar/ouster-ros/blob/d48c331cfa299d3c85c5182bb38dbaf746965a04/launch/record.launch#L26

JACKLiuDay, 2022-12-02:
> This is the line you're looking for:
> 
> https://github.com/ouster-lidar/ouster-ros/blob/d48c331cfa299d3c85c5182bb38dbaf746965a04/launch/record.launch#L26

Thank you for your kind reply. I just found my ouster-ros version is old.  I will try the new version. Thank you again.

juliangaal, 2022-12-04:
Yeah, the driver went through some large restructuring, but (in my limited experience) works very well

mahammadirfan, 2023-12-13:
@YushengWHU 
Hi, I am using ouster os1 ...can you please share your yml file ? thanks

YushengWHU, 2023-12-14:
> @YushengWHU Hi, I am using ouster os1 ...can you please share your yml file ? thanks

sorry, its really a long time ago, and I dont have that yaml file now, even my os1 is broken.

ZhangYaoll, 2024-05-15:
> > IchliebeYe
> 
> 我微信就这个啊。。IchliebeYe

加不上。你是微信国际版的吗